moonlight-stream/moonlight-embedded #438: Controller aims straight up on games

TheConsciousness, 2016-11-19:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.1.2.31 (latest 11/18/16)
**Moonlight Embedded version:** 2.2.2
**Moonlight Embedded running on:** Raspberry Pi 3

**Output of Moonlight Embedded:** 
Normal output

**What is the expected result?**
Cursor disappears and controller provides the ability to aim in Steam first person shooter games. 

**What happens instead of that?**
Cursor remains in middle of screen and aim always moves straight up. My character looks to the sky.

7hunderbug, 2016-11-19:
Are you using a carrot as the controller? If so, are you using the default carrot cake recipe? If you are and the problem still persists you might need more cinnamon otherwise you should try a different recipe; how about carrot salad?

irtimmer, 2017-03-18:
You need to map the controller to fix issues with input

closed 2017-03-18 by irtimmer