moonlight-stream/moonlight-embedded #440: Video Ping: send() failed: 101

royeectu, 2016-11-24:
Please provide the following info.

**_NVidia Geforce Experience version: 3.1.2.31 (protocol version 7)
**Moonlight Embedded version: 2.2.2 (EMBEDDED;PI)
**Moonlight Embedded running on:** Raspberry Pi 3

**Output of Moonlight Embedded:**
Video Ping: send() failed: 101
Stopping input stream...done
Stopping audio stream...Control stream connection failed
Loss Stats: Transaction failed: 101

**What is the expected result?**
That my Steam games won't crash randomly with this error message
**What happens instead of that?**
Every Steam game that I have crashes suddenly with this error message

royeectu, 2016-11-25:
Anyone? Please, I have to solve this issue

Spamm00r, 2017-10-16:
I think we have the same issue.

WHat Wifi dongle do you use?

geekycow, 2018-03-24:
I'm now seeing this on the latest Raspbian Stretch on a Pi 3 B+ over Ethernet (1Gbps) but not on a Pi 2 over Ethernet (100Mbps.)

geekycow, 2018-03-24:
Ack. Ignore me. Looking at syslog this is coinciding with eth0 carrier lost events, which really should NOT be happening on decent quality wired network. (And didn't with the Pi 2.)

closed 2019-01-14 by royeectu