moonlight-stream/moonlight-embedded #443: Provide PTS clock information for syncronization

OtherCrashOverride, 2016-12-07:
Please provide the following info.

**_NVidia Geforce Experience version:_** : ALL
**Moonlight Embedded version:** : ALL
**Moonlight Embedded running on:** : ALL

**Output of Moonlight Embedded:** N/A

**What is the expected result?**
Video and audio synchronize with the PC host.

**What happens instead of that?**
Video and audio drift out of sync over time.

With any live media source, a reference clock is necessary to compensate for clock drift.  This information needs to be provided to the video/audio decoders to maintain synchronization (frame dropping/stretching).  Currently, moonlight-embedded does not appear to expose this information to decoders.

cgutman, 2016-12-07:
This needs to be done with care. Under no circumstances do we want to delay the video even a single frame. Video latency is king here. Dropping audio frames to catch up with the video is probably okay but extending video frames is definitely not.

However, the majority of A/V sync issues that I see on the Pi is audio getting behind video (due to the CPU overhead involved in doing software Opus decoding). In this case, we could do a better job dropping audio samples in a calculated manner (rather than all at once when the queue reaches the threshold).

OtherCrashOverride, 2016-12-07:
Currently, frame stretching happens as a side-effect.  The client codec is starved of data and has to await the next frame.

The current behavior is that all codecs run asynchronous until either starved of data or a buffer overflows.  This then causes a synchronization as a side-effect.  The goal is to make this deterministic by surfacing the media reference clock provided by the host in the media stream and exposing it in the client API to make use of should it choose.

Although the host may produce 60fps and the client may display at 60fps, The clocks between host and client will never be perfectly in sync.  This means there is always a relative time delta that accumulates.  If the delta favors the host, then the client will need to drop data. If, instead, it favors the client, the client will need to stretch data.

irtimmer, 2017-03-18:
Even if host and client are not in sync it won't happen very often. Probably only once in x minutes at max and a single frame extra isn't very noticeable. In the case a frame is dropped it is even less noticeable as it got replaced by another frame.

closed 2017-03-18 by irtimmer