moonlight-stream/moonlight-embedded #446: no video on geforce 3.2.0.96

philocritus, 2016-12-14:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.2.0.96
**Moonlight Embedded version:** 2.2.2
**Moonlight Embedded running on:** Raspberry Pi

**Output of Moonlight Embedded:**  loop message "Waiting for IDR frame"

**What is the expected result?** Working game video stream

**What happens instead of that?** server works OK but moonlight shows only the message "Waiting for IDR frame" on loop

This seems to already be patched on the main moonlight version. Only a embbebed version release should be needed:
https://github.com/moonlight-stream/moonlight-common-c/commit/f1adbc1e6bd68f8abf6855308e176d2bfc906693

schif1, 2016-12-14:
Same issue for me on Raspberry Pi 3.

tocco93, 2016-12-15:
same issue  on my pi 2

mschewie5, 2016-12-15:
Hallo same to me since two days
geforce experience: 3.2.0.96 installed today
moonlight: 2.2.2
raspberry pi 2: fully updated, opengl is off
console shows only the loop message, but control and audio working well.

1031982, 2016-12-16:
I have this as well, just updated my GPU and thought that might be it. I had a 650 ans went to a 960.
GeForce experience 3.2.0.96
Moonlight Embedded - 2.2.2
Raspberry Pi 3 with RetroPi 4.1 - I updated sometime in November.

Akensy, 2016-12-16:
Same here.
Was working fine 2 days ago (previous GFE version).
Raspberry Pi3 Model B fully updated.

irtimmer, 2016-12-16:
Until the next version of Moonlight Embedded is released, the simplest way to get it working again is to downgrade to version 2.2.1

http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/moonlight-embedded_2.2.1-1jessie1_armhf.deb
http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/moonlight-embedded_2.2.1-1wheezy1_armhf.deb

philocritus, 2016-12-16:
I can confirm downgrading to 2.2.1 works fine.
 I suspected it and was about to try it when irtimmer commented 👍

gnarlyelesdee, 2016-12-16:
Can anyone posts steps on downgrading?  Can I just install over 2.2.2?

philocritus, 2016-12-16:
do the following on your command line:
sudo dpkg --install DEB_FILE.deb

no need to uninstall previous version

1031982, 2016-12-16:
I to have it working on 2.2.1.
I went through and un-installed 2.2.2 before installing 2.2.1. Actually, I purged it. I prefer to have a clean install to prevent issues, plus it's easy to set up.
In my RetroPi setup, I added Steam with scripts to call on opening moonlight with a controller of choice. All the map files are in a sub folder I made in Home. This way the only thing I should need to do is pair it with my PC. Usually it keeps the pairing between versions, even when I purge it, so it's real simple.

Shoehart, 2016-12-17:
Is it possible for some one to share compiled version of 2.2.1 for a Raspi3 desperate user?

leizzer, 2016-12-18:
the solution of downgrade works perfectly

tocco93, 2016-12-18:
sudo command dont work on recalbox

tocco93, 2016-12-18:
other easy solution is to unistall GFE  3.2.0.96 and install GFE 3.1.2.31.
link:
http://www.filehorse.com/download-nvidia-geforce-experience/old-versions/

figpicker, 2016-12-19:
I am also new to moonlight embedded and raspberry pi. Could someone please help explain how to downgrade to 2.2.1?

Beric77, 2016-12-19:
@tocco93 This is not a solution. After GFE will update to the new version again.

If you try to block conection in the firewall, then login doesnt work and you cant use the application...

Now, i think the only way to can use moonlight is the downgrade to 2.2.1 version. i try it and it works.

tocco93, 2016-12-19:
With recalbox I can't downgrade so i do what I say before. It work

georgcmef, 2016-12-19:
Hey all, I built commit 51a602f7edad5c0006b0f1eb0043ccfc0518b568 and its working fine. Is it fine if I use Checkinstall to create a .deb and share it here? This is my first time making a .deb file so forgive me if it doesn't work for you guys. Its only for Jessie

[https://packagecloud.io/georgc/MyRepo/packages/raspbian/jessie/build_20161219-1_armhf.deb](url)

irtimmer, 2016-12-20:
New version with fix is released

closed 2016-12-20 by irtimmer

Beric77, 2016-12-21:
@irtimmer New version GFE or Moonlight ?

irtimmer, 2016-12-21:
@Beric77 https://github.com/irtimmer/moonlight-embedded/releases