moonlight-stream/moonlight-embedded #447: Waiting for IDR Frame

gnarlyelesdee, 2016-12-16:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.2.0.96
**Moonlight Embedded version:** embedded 2.2.2
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._ pi3b

**Output of Moonlight Embedded:**default

**What is the expected result?**stream 

**What happens instead of that?** Opens steam on PC and I am able to control using keyboard, on pi screen does non stop waiting for IDR frame until I close steam on PC

digix77, 2016-12-17:
I've the same version and the same problem.

4eMaLo, 2016-12-17:
Apparently, NVidia changed their protocol in GeForce Experience 3.2:
[Commit Message for moonlight-common-c](https://github.com/moonlight-stream/moonlight-common-c/commit/f1adbc1e6bd68f8abf6855308e176d2bfc906693)

digix77, 2016-12-18:
I compiled the cloned git-repository myself, too, and it works also.

spookytay, 2016-12-18:
how do you go about doing that, I've been trying to figure it out, but cannot get it to work

irtimmer, 2016-12-18:
Duplicate of #446

closed 2016-12-18 by irtimmer