moonlight-stream/moonlight-embedded #449: Connecting to Linux steam host

ant-thomas, 2016-12-23:
This may be a linux/ubuntu Steam issue rather than moonlight, but some advice would be great.

Currently running Steam on Ubuntu, fully updated with in-home streaming enabled - moonlight can't find the server or connect directly if I give it the IP.

I'm running moonlight-embedded on a Raspberry Pi 1 Model B.

Is this a known issue? Is there something specific that needs enabling on Steam on Ubuntu to get moonlight to find the server?

Moonlight Embedded 2.2.2 (EMBEDDED;PI)
Connect to 192.168.1.227...
Can't connect to server 192.168.1.227

If this isn't relevant to moonlight, and should be directed towards the Steam forums let me know.

irtimmer, 2016-12-23:
I think you need to re-read the requirements

closed 2016-12-23 by irtimmer

ant-thomas, 2016-12-23:
I assumed it supported Steam In-home streaming from Linux too. I was wrong, sorry.