moonlight-stream/moonlight-embedded #450: It's not possible to quit Moonlight

Frolus, 2016-12-25:
**NVidia Geforce Experience version:**  GFE 3.2.0.96
**Moonlight Embedded version:** 2.2.2
**Moonlight Embedded running on:** Raspberry Pi2 (RetroPie)

**Output of Moonlight Embedded:**
after starting the game not in Steam (or explorer.exe) via the Moonlight no way to exit the Moonlight back to EmulationStation and stop from the gamepad (Dualshock3) stream without SSH

**What is the expected result?**
Availible exit from MoonLight by hotkey on gamepad

**What happens instead of that?**
the only way at the moment that I found is to add all my games to Steam lib. and only run through it, as when you exit Steam translation stop.

jasonridesabike, 2017-01-10:
From the Wiki:
Use Ctrl+Alt+Shift+Q to exit streaming session
https://github.com/irtimmer/moonlight-embedded/wiki/Usage

irtimmer, 2017-03-18:
Duplicate of #317

closed 2017-03-18 by irtimmer