moonlight-stream/moonlight-embedded #452: How to remap an controller

ItsVince1234, 2016-12-28:
i want to remap my controller and set the file but id dont know how to do it can you help me please?

tapestreamer, 2017-01-09:
Use the mapping option to create your own map file on your client.

Use `moonlight map controller.map` and press the buttons to map them.

Make sure to link to the map file, i.e.:
`moonlight stream -720 -30fps -mapping /home/pi/moonlight/controller.map`

jasonridesabike, 2017-01-10:
Note that the moonlight map whatever.map saves the file to whatever directory you run it from and you don't have to include the path to the map file with -mapping if it's in the same directory from which you run Moonlight. Useful for configuring Moonlight to work well with EmulationStation.

closed 2017-03-18 by irtimmer