moonlight-stream/moonlight-embedded #453: error while loading shared libraries: libgamestream.so.0: cannot open shared object file: No such file or directory

GrapeJamMan, 2016-12-29:
Just built from source. When installing dependencies, apt-get can't find these two packages:
E: Unable to locate package rbp-userland-osmc
E: Unable to locate package rbp-userland-dev-osmc

This is after an apt-get update so it's current. I left them out and cmake worked anyway, and so did make and make install.

Not sure what I'm doing wrong:

Please provide the following info.

**_NVidia Geforce Experience version:_**
3.2.0.96

**Moonlight Embedded version:**
2.2.3

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Raspberry Pi 3

**Output of Moonlight Embedded:**
moonlight: error while loading shared libraries: libgamestream.so.0: cannot open shared object file: No such file or directory

**What is the expected result?**
Create mapping file for gamepad

**What happens instead of that?**
Error message

ergarcia516, 2017-01-20:
Encountering the same error. Any movement on this?

irtimmer, 2017-02-23:
Duplicate of #210

closed 2017-02-23 by irtimmer