moonlight-stream/moonlight-embedded #456: Cannot install proper version (2.2.3)

jeremyabraham7, 2016-12-31:
This may be my noobness but im struggling with this issue https://github.com/irtimmer/moonlight-embedded/issues/446 and cannot seem to get Moonlight 2.2.3 installed to solve it using apt-get it shows version 2.2.2. I tried both updating using `apt-get` and wget'ing the deb file directly. I've also tried a downgrade to 2.2.1 which shows version 2.2.0 

**_NVidia Geforce Experience version:_** 3.2.0.96 (protocol version 7)
**Moonlight Embedded version:** 2.2.2 (?)
**Moonlight Embedded running on:**  Raspberry Pi 3/RetroPie

**Output of Moonlight Embedded:**
pi@retropie:~ $ sudo apt-get install moonlight-embedded
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  moonlight-embedded
0 upgraded, 1 newly installed, 0 to remove and 49 not upgraded.
Need to get 0 B/83.3 kB of archives.
After this operation, 298 kB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  moonlight-embedded
Install these packages without verification? [y/N] y
Selecting previously unselected package moonlight-embedded.
(Reading database ... 84531 files and directories currently installed.)
Preparing to unpack .../moonlight-embedded_2.2.3-1jessie1_armhf.deb ...
Unpacking moonlight-embedded (2.2.3-1jessie1) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up moonlight-embedded (2.2.3-1jessie1) ...
Processing triggers for libc-bin (2.19-18+deb8u6) ...

**What is the expected result?** `moonlight` should show version 2.2.3 when installed via `apt-get`

**What happens instead of that?** shows version 2.2.2 despite the install clearing showing it's downloading the 2.2.3 deb. Attempting a downgrade to 2.2.1 outputs version 2.2.0 when calling `moonlight` - let me know what other logs/file/info I can provide (and how to get them, - pretty new to this)

Thanks for the help!

jasonridesabike, 2017-01-05:
@jeremyabraham7 It looks like the PPA just hasn't been updated with the latest version; logged on myself to report that issue. I'm just going to compile from source and you can too.

just: `sudo apt-get install git`
`mkdir ~/Moonlight`
`cd ~/Moonlight`
`git clone https://github.com/irtimmer/moonlight-embedded.git .`

Then follow all the instructions here: https://github.com/irtimmer/moonlight-embedded/wiki/Compilation

Make sure to apt-get install all those dependencies before getting started. You can install multiple packages at once by separating them with a space: `sudo apt-get install libssl-dev libopus-dev another-package etc`. You'll want to run all the compile commands from that Moonlight folder we created, but if you followed instructions you'll already be there.

rmogedano, 2017-01-10:
O have the same problem, help please.

Soldia1138, 2017-01-10:
Solution is simple: The Version number is wrong (still 2.2.2): https://github.com/irtimmer/moonlight-embedded/blob/master/CMakeLists.txt#L6

jasonridesabike, 2017-01-10:
haha ya, I noticed that after compiling and just went with the PPA version. Thanks!

dmnc-net, 2017-03-13:
@irtimmer @Soldia1138 https://github.com/irtimmer/moonlight-embedded/blob/master/CMakeLists.txt#L8

closed 2017-03-18 by irtimmer