moonlight-stream/moonlight-embedded #457: Can't use xbox controller on host PC

Zelaf, 2017-01-06:
Hello! I'm trying to control the games on a Xbox controller that is connected to my host PC and not the Raspberry Pi itself. When I googled I found a possible solution that could fix the issue by renaming the `rxinput.dll` file to `rxinput.dll.old` but it did not work. If there is an updated way to do this or a workaround that would be great.

```
Please provide the following info.

**_NVidia Geforce Experience version: 3.2.0.96**
**Moonlight Embedded version: 2.2.2 (Repo version)**
**Moonlight Embedded running on: Raspberry Pi B+** 

**Output of Moonlight Embedded:

pi@raspberrypi:~ $ moonlight stream -audio local
Moonlight Embedded 2.2.2 (EMBEDDED;PI)
Searching for server...
Connect to 192.168.1.91...
NVIDIA GeForce GTX 650, GFE 3.2.0.96 (protocol version 7)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done

**

**What is the expected result?: Controlling the games and steam through the Xbox controller that is connected to the host computer. **

**What happens instead of that?: I can navigate Steam just fine but when I start a game the controller just stops. No input at all being detected through the controller and I have to manually quit the game by either killing it or stopping it.**

```

When I have the xbox controller plugged in to the RPi the controller works but the mapping is wrong. I don't want to have the controller plugged in to my Pi in the first place since I'm trying to get a bit of a special setup.

irtimmer, 2017-01-06:
https://github.com/moonlight-stream/moonlight-docs/wiki/Setup-Guide#using-a-gamepad-connected-to-the-pc-instead-of-the-streaming-device

closed 2017-01-06 by irtimmer

Zelaf, 2017-01-06:
Why didn't I think of looking through the moonlight-stream documentation? Well lesson learned. 

Thanks.