moonlight-stream/moonlight-embedded #459: Xbox one controller not responding

leonardoholberg, 2017-01-10:
**NVidia Geforce Experience 3.2.2.49

Moonlight Embedded 2.2.2

Moonlight Embedded running on Raspberry Pi 2B**

my problem is, that when I connect my xbox one wireless controller to my raspberry pi, it shows up in the list of input devices (when typing in cat /proc/bus/input/devices) but when I start streaming or try to map it doesn't respond, and the xbox logo dosnt glow, do I have to install drivers? I use a 2 amp power adapter do I dont think its that

thanks -Leonardo

jasonridesabike, 2017-01-10:
Had the same problem ended up using Retropie as it includes a preinstalled kernel driver for xboxone controllers. Getting them installed by any other route was just taking too much time. In any case I imagine drivers are out of scope for Moonlight.

noxx2015, 2017-02-04:
is it also for xbox360?

i try this, but i get

Klone nach '/usr/src/xpad-0.4'...
remote: Counting objects: 249, done.
remote: Total 249 (delta 0), reused 0 (delta 0), pack-reused 249
Empfange Objekte: 100% (249/249), 132.18 KiB | 0 bytes/s, Fertig.
Löse Unterschiede auf: 100% (152/152), Fertig.
Prüfe Konnektivität... Fertig.
pi@raspberrypi:~$ sudo dkms install -m xpad -v 0.4

Creating symlink /var/lib/dkms/xpad/0.4/source ->
                 /usr/src/xpad-0.4

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.4.34-v7+ cannot be found at
/lib/modules/4.4.34-v7+/build or /lib/modules/4.4.34-v7+/source.

irtimmer, 2017-03-18:
Closed as this isn't a bug in Moonlight Embedded

closed 2017-03-18 by irtimmer