moonlight-stream/moonlight-embedded #464: Moonlight exits after loading, but host works properly.

1031982, 2017-01-16:
Nothing has changed in my setup on either the Raspberry Pi or my PC.
After Moonlight closes down, my PC keeps the stream running as if it was working properly. Even choosing the power icon in Steam Big Picture mode shows two options, "Stop Streaming", and "Shutdown".

NVidia Geforce Experience version: 3.2.2.49
Moonlight Embedded version: 2.2.3
Moonlight Embedded running on: Raspberry Pi 3

From command line, after starting input stream ... done, I have this:
Control Stream received disconnect event
Loss Stats: Transaction failed: 11

Then it shows the standard stopping lines.

It was expected to load the Steam Big Screen Mode, instead it closes out to either Emulation Station or command line. Whichever moonlight was launched from.

I have included nvstreamer log.
[NvStreamerCurrent.txt](https://github.com/irtimmer/moonlight-embedded/files/707428/NvStreamerCurrent.txt)

irtimmer, 2017-01-16:
It looks like you have some third party software running in the background which is incompatible with Moonlight. The stream can't be started because some third-party software is showing protected content on the screen (which can't be streamed) according to the logs (NVFBC_ERROR_PROTECTED_CONTENT).

1031982, 2017-01-16:
The only thing that has changed on my system is that I have removed and some games, and installed some new ones.
Heck, I can give you a list of everything running on my system. I actually have less running on it now then a few weeks ago, and it worked fine then.
All my most recent installs are LEGO Marvel Avengers, VUDO To Go was updated, and AOMEI Partition Assistant. I doubt any of those would cause a problem.

galvatrondeva, 2017-01-16:
Do you have TeamViewer installed by any chance? I've had issues with the QuickConnect button in the past and disabling that particular function helped.

1031982, 2017-01-16:
Nope, but now that I think about it, could installing a HVAC H265codec cause a problem? I installed LAV Filters.

1031982, 2017-01-16:
Strange, I rebooted my system and reinstalled GFE and now it's working fine.

closed 2017-01-16 by 1031982