moonlight-stream/moonlight-embedded #473: XBOX 360 Rumble

noxx2015, 2017-02-16:
i read here, that Rumble isnt supported by moonlight. 
in nvidia forum i read this here
https://forums.geforce.com/default/topic/986769/gamestream-rumble-feedback-on-original-shield-controller-/?offset=6

is it possible to use rumble with a xbox 360 controller?

irtimmer, 2017-03-18:
This isn't yet supported as the details of the protocol for rumbling are currently unknown

noxx2015, 2017-03-18:
thx, i must switch to SteamOS after Moonlight will not run on LibreELEC (x64). not so nice as Moonlight, but better as nothing.

posti85, 2018-11-26:
Is it still unsupported? Just to know. Best regards and thank you very much for this amazing project.

posti85, 2019-02-12:
> This isn't yet supported as the details of the protocol for rumbling are currently unknown

Now there are news on this :) https://github.com/moonlight-stream/moonlight-common-c/commit/fa86e76978e4376afe94b13b64e94b4846f85f02

posti85, 2019-02-19:
Hi @irtimmer, I've tested the project with the lastest commits about rumble support, but in my xbox 360 wireless controller the rumble feature still doesn't work. That make sense?

posti85, 2019-02-20:
Can confirm with `sdl2-jstest -r 0` (https://gitlab.com/sdl-jstest/sdl-jstest) that vibration is working properly with the current xpad driver. I've also seen 'Initial attempt to implement rumble for evdev' in a commit message, so I think you are still developing the feature.

posti85, 2019-04-29:
I've been playing with my recent fixes about the rumble support (https://github.com/irtimmer/moonlight-embedded/pull/740) and for me it is working in xbox 360 controllers.

cgutman, 2021-07-24:
Fixed

closed 2021-07-24 by cgutman