moonlight-stream/moonlight-embedded #478: Moonlight on Retropie won't display fullscreen

nareau, 2017-03-02:
**_NVidia Geforce Experience version:_**
3.4.0.70

**Moonlight Embedded version:**
2.1.0

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Raspberry Pi 3, Retropie

**What is the expected result?**
I expect steam to fill the screen.

**What happens instead of that?**
Only the middle 2/3rds of the screen are filled. Black bars with white text can be seen "behind" the Moonlight stream at the top and bottom of the TV. Both my PC's monitor, and my TV are natively 1920x1080.

This is the command I'm running to start Moonlight:
moonlight stream -1080 -30fps  -mapping /opt/retropie/configs/moonlight/controller.map 10.0.0.10

[image]

I've tried changing the resolution on my monitor, and the resolution within Steam. I've also tried adding  **-height1080** to the command. Nothing seems to have any effect. I've tried on another 1080p display, and got the same thing. I was able to get fullscreen working out of the box when I had it connected to a 7" touchscreen.

Laephis, 2017-03-04:
I'm having the same problem.  I've been using Moonlight on Retropie on and off for a while and I want to say this problem popped up a few months ago.  Just getting around to trying to fix it..

brainfrz82, 2017-03-12:
Same issue here, ive been trying some remote desktop apps for the rpi 3 to connect to my pc. At first, up until a few hours ago moonlight was working fine. After settling for moonlight i decided to uninstall all the other remote desktop apps i tried. Nomachine, freerdp and vnc client were uninstalled. After a reboot moonlight stopped working full screen. I havent been able to replicate the original working setting, even after reinstalling all the other tools.

irtimmer, 2017-03-18:
It looks like the aspect ratio is incorrectly set or detected on the Raspberry Pi. But it looks like there are some functions to change the aspect ratio in code so maybe I will add them in the next release.

Black4Key, 2017-04-01:
Same problem for me, but neither the windows control panel nor nvidia control panel work for me, every time the stream start it adjust itself as it want.
(I'm using moonlight embedded version 2.3.0 on a rasperry pi 3 model B+ with raspbian as OS)

tbergeron, 2017-04-13:
Currently having the same problem. Same config as the fellow commenter above me ^^

alfalfa1978, 2017-04-16:
Fullscreen works fine for me on RPi 3 with RetroPie.
@tbergeron The scaling mode in my NV control panel is still set to "aspect ratio", but I got no problems with fullscreen.

I've only got a problem using CEMU (WiiU Emulator) - in fullscreen mode the screen freezes while streaming (emulator still running), but if you switch to the emulator's windowed mode, it works until you switch back to fullscreen.
Several people have the problem but nobody knows if it's a problem of CEMU or Moonlight Embedded.

cgutman, 2017-04-16:
@alfalfa1978 It's neither CEMU or Moonlight's problem. Nvidia's frame buffer capture doesn't correctly handle certain full-screen rendering methods. If one of these applications is the only visible window, the display won't update. Chrome, Edge, and some other UWP apps have this issue when displaying video in full-screen. This is the first I've heard of a game triggering it, but it doesn't really surprise me.

alfalfa1978, 2017-04-16:
Thank you for your explanation!
I hope this will by fixed anyhow.
 
Here’s the link to the regarding reddit thread:
https://www.reddit.com/r/cemu/comments/5qayto/issue_with_moonlight_and_nvidia_gamestream/
 
Greetings from Berlin
Thomas

alfalfa1978, 2017-04-17:
Hey Cameron,
 
A reddit user just posted:
https://www.reddit.com/r/cemu/comments/5qayto/issue_with_moonlight_and_nvidia_gamestream/dgcik54/
 
It works fine for me!
Perhaps it also fixes other full-screen issues.

irtimmer, 2017-05-27:
Duplicate of #364

closed 2017-05-27 by irtimmer