moonlight-stream/moonlight-embedded #481: Connection cut off after I hit a button

alfalfa1978, 2017-03-18:
Please provide the following info.

**_NVidia Geforce Experience version: 3.4.0.70
**Moonlight Embedded version: 2.3.0
**Moonlight Embedded running on: Raspberry Pi 3 via shortcut in RetroPie
**Output of Moonlight Embedded: segmentation fault, several numbers displayed

**What is the expected result?**
running as usual
**What happens instead of that?**
connection is cut off

I updated to moonlight embedded 2.3.0 today.

When I start streaming, it does connect an opens the startscreen of steam.
It works and does play the music until I hit any button or the directional pad (xbox360 wireless controller) and then the connection is cut off, saying "segmentation fault".

With the previous version I hadn't any problems at all.

irtimmer, 2017-03-19:
Moonlight Embedded 2.3.0 works fine with controllers on ArchLinux ARM. So probably there is something wrong with the Debian build or your setup. But there is not much to go on if you only provide a very general error string "Segmentation Fault"

alfalfa1978, 2017-03-19:
I'm sorry, don't know where to find a logfile for further Information.
If I start moonlight, steam big picture mode appears and plays the background Music.
If I hit any button oder directions on the xbox360 wireless controller, streaming stops and then I can read "851 segmentation fault" or "868 segmentation fault" or "886 segmentation fault" - the number changes everytime I try it again.

If I move the thumbsticks or hit buttons while moonlight is starting (RTSP handshake, initializing and starting the streams) i get the notice: "Error - cannot keep up".

I use RetroPie 4.1.26 (premade image for the pi) and start moonlight out of the RetroPie Surface using a shortcut with the following commands after the IP: -1080 -60 fps -Bitrate -surround -mapping /(location of my mapping file) -app Steam

With all previous versions of Moonlight the controller and the streaming worked without any problems.
In RetroPie itself and all the emulators there are also no problems.

If I start Moonlight without the "-mapping" command, it also stops streaming when I hit any button, saying "unmapped button".

Did I anything wrong updating from existing Moonlight 2.2.3 to Moonlight 2.3.0 just using: "sudo apt-get update" followed by "sudo apt-get install moonlight-embedded"?

irtimmer, 2017-03-22:
You can try to run `sudo apt-get upgrade` to do a full system upgrade. After installing Raspbian I can only confirm the Debian build is still working fine with gamepads.

alfalfa1978, 2017-03-24:
Ok, thank you.
In order to upgrade, I first had to run sudo apt-get autoremove. Otherwise it did nothing, only saying, that there are packages no longer required.
After autoremove I ran sudo apt-get upgrade and then it did the upgrade.
Then I ran sudo apt-get update.

But in the end I got the same issue... it crashes back to the RetroPie menu as soon as I hit a button / direction...
I noticed today that after the the segmentation fault is displayed, it says that moonlight 2.2.2 was started.
I'll try now installing moonlight 2.3.0 by downloading and compiling.
Perhaps there is a problem by just doing the update.

irtimmer, 2017-03-25:
Older versions of Moonlight-embedded can just be obtained from http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/ and installed with `dpkg -i "deb filename"` There are some obstacles with compiling because of the old versions Debian is using. Therefore libcec is disabled in my builds for Debian and I'm using a patch from debian/patches/install.patch (in the raspbian/jessie branch) to fix issues with compiling and cmake.

alfalfa1978, 2017-03-27:
How do I install the patch?
I'm to stupid, sorry.
Somehow now I cannot connect anymore. "Server not found".
I whish I simply could uninstall moonlight-embedded completeliy and then reinstall it, but I don't know how to do.
After "sudo apt-get remove moonlight-embedded" it is still there, because it knows the command "moonlight".

irtimmer, 2017-03-27:
A patch can be applied to the sourcecode using `patch -p1 < "filename patch"`. But since moonlight is still available after removal you probably had installed multiple versions of Moonlight Embedded at the same time. For example a package version and a self-compiled version. Which explains why you get a segmentation fault as the code for sending buttons has changed and probably a older version of an helper library got loaded. You have to remove all libgamestream.so and libmoonlight-common.so from your system before reinstalling Moonlight Embedded.

closed 2017-03-27 by irtimmer