moonlight-stream/moonlight-embedded #487: EVIOCGRAB failed with error 16

csmith37, 2017-04-03:
**NVidia Geforce Experience version:** 3.4.0.70
**Moonlight Embedded version:** 2.3.0
**Moonlight Embedded running on:** Raspberry Pi 3 - OSMC

**Output of Moonlight Embedded:**
osmc@osmc:~$ moonlight stream 192.168.0.107
Moonlight Embedded 2.3.0 (EMBEDDED;PI)
Connect to 192.168.0.107...
NVIDIA GeForce GTX 980 Ti x2, GFE 3.4.0.70 (protocol version 7)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
EVIOCGRAB failed with error 16
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Terminating RTSP handshake...done
Cleaning up platform...done
osmc@osmc:~$

**What is the expected result?**
Stream should start

**What happens instead of that?**
Stream immediately crashes

I have viewed #422 and the issues linked to it, but their issue appears to have changed after rebooting. Mine has not.

Attempted to do a clean reinstall of NVIDIA drivers and GFE.
Attempted to reinstall moonlight-embedded. 
No luck.

irtimmer, 2017-04-03:
The error "EVIOCGRAB failed with error 16" is unrelated to the disconnecting issue and is a rights issue of your input devices trying to disable keyboard input to the console during the usage of Moonlight. Ru with sudo or change permissions of /dev/input/*.

"Loss Stats: Transaction failed: 11" is a issue with your host and most likely a compatibility issue of GFE with other installed software. Check also https://github.com/moonlight-stream/moonlight-docs/wiki/Troubleshooting . If this doesn't resolve your issue please upload the files in C:\ProgramData\NVIDIA Corporation\NvStream so this issue can be further debugged.

csmith37, 2017-04-03:
I have changed the permissions of /dev, /dev/input, and /dev/input/* to 777 and rebooted the PI, but I still see the "EVIOCGRAB failed with error 16" message when starting moonlight stream.

I had TeamViewer installed but not running (moonlight was working previously even when I had it installed). I uninstalled it and rebooted, but still the "Loss Stats: Transaction failed: 11" message occurred. No other software I have appears on the incompatibility list.

I did an uninstall from safe mode and clean reinstall of my nvidia drivers and GFE again to be sure. No luck. 

[NvStreamSrv(x64).zip](https://github.com/irtimmer/moonlight-embedded/files/891528/NvStreamSrv.x64.zip)
[NvStreamSrv(x86).zip](https://github.com/irtimmer/moonlight-embedded/files/891529/NvStreamSrv.x86.zip)

csmith37, 2017-04-03:
So by starting with a fresh installation of OSMC and installing moonlight embedded on the new system, it started working. This brings me to another question.

Is there a guide to fully uninstalling all remnants of moonlight embedded. I tried to uninstall and reinstall on my existing OSMC installation, but the issue still persisted. What directories and files would I need to remove to get rid of any potentially bad configuration?

irtimmer, 2017-04-04:
Manual changes of the /dev/input/* rights will be reverted after a reboot. The correct way to give a user access to the input devices depends on the used distribution.

Moonlight Embedded only saves some key data for pairing to $XDG_CACHE_DIR/moonlight or ~/.cache/moonlight But that shouldn't result in many problems. Although there are people installing multiple versions (for example from source and from a package) which can result to unexpected behaviour. 

_p.s. Log files are in **ProgramData** not in **Program Files** ;)_

closed 2017-04-04 by irtimmer