moonlight-stream/moonlight-embedded #488: Should this version work on a Gigabyte Brix (x64) and LibreElec ?

MopheusDG, 2017-04-03:
Nvidia GE version 3.4.0.70
Moonlight Embedded - Latest
Running on Gigabyte Brix J1900 and LibreElec 8.0.1

I'm using the official LE addon for Moonlight and I only get a black screen. Worked once but never again and I'm not sure why. This version runs fine on LE 8.0.1 on RPi3 but not on my x64 Mini PC, but I'm not really sure if it's supposed to.

By the way, is there any chance to make an official LE addon but done by the Moonlight team ? I think LE is great for Moonlight, at least on the RPi3 works amazing.

Thanks.

irtimmer, 2017-04-04:
Theoretically there is a software decoder implemented in Moonlight Embedded which should be usable on a x64 system. But it depends on how Moonlight is integrated into LE if it works or not. Chances a official LE addon is released by the "Moonlight team" are very small.

_p.s. if there is a bug preventing Moonlight Embedded from running you can open an issue with the required debug information._

closed 2017-04-04 by irtimmer

CvH, 2017-04-04:
We have no one with the needed hardware to test this at all, all I know is that there are problems with vaapi (green screen) that it won't work at intel HW. It is unlikely to get this working from LE side (at least atm).

irtimmer, 2017-04-04:
@CvH Vaapi isn't supported by Moonlight Embedded and will only be 'used' if a crappy vaapi-to-vdpau wrapper is installed and `-forcehw` is given as argument (will should never be used) or an older version of Moonlight Embedded is used.