moonlight-stream/moonlight-embedded #489: No audio from center channel with 5.1

abrenoch, 2017-04-06:
**_NVidia Geforce Experience version:_**
3.4.0.70

**Moonlight Embedded version:**
8.1.105

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Raspberry pi 2 with libreelec 8

**Output of Moonlight Embedded:**
N/A

**What is the expected result?**
With 5.1 audio enabled I am expecting the output to all the speakers

**What happens instead of that?**
All channels work except the center

I have my pi connected to my TV which connect to my onkyo receiver over ARC and all other 5.1 works fine for me, but moonlight does not seem to send any audio to the center channel. typically when I start some video with 5.1 my receiver will display the codec being used, but that does not happen with moonlight.

If there is anything I can provide to help with this please let me know!

abrenoch, 2017-04-11:
Just an update here -  I think the surround sound is messed up in general and being output as stereo.. I think I was only under the impression the center was not working because of the behavior of my reciever (I later noticed some ambient noise coming from it).

irtimmer, 2017-04-11:
I think your setup is just messed up. You're trying to send uncompressed PCM 5.1 over ARC to your receiver. See also http://www.rtings.com/tv/tests/inputs/5-1-surround-audio-passthrough

closed 2017-04-11 by irtimmer

abrenoch, 2017-04-11:
So it sounds like I should try again using eith an optical cable or by hooking the pi directly to the receiver?

The jist I took away from the link you provided is that no TV passes PCM as 5.1 to the receiver, correct? 

That would explain why the 5.1 movies I have work fine, because they use a codec like DD or DTS assuming I understood correctly!

abrenoch, 2017-04-12:
Just wanted to report that changing the way things are hooked up did work for me! I just hooked the pi directly to the receiver rather than to the TV and using ARC to pass the audio to the receiver. Doing so also fixed a couple other minor problems I was experiencing (like the audio not always passing through and the TV remote acting unresponsive in general).. The only reason I had hooked it up that way is that I had assumed the TV remote would not work with kodi if the pi was not hooked up to the TV, but I was incorrect in that assumption as it works better than it did before!

Thanks for the tip, I knew it had to be something dumb like that!