moonlight-stream/moonlight-embedded #490: Issue #100 rises again: still having keyboard issues with non-alphanumeric keys

maxellnormalbias, 2017-04-13:
**_NVidia Geforce Experience version:_** 3.5.0.70
**Moonlight Embedded version:** 2.3.0
**Moonlight Embedded running on:** Thinkpad T420

**Output of Moonlight Embedded:**
Shell output looks fine, video displays as expected

**What is the expected result?**
Being able to type the keys ,./][;:' etc

**What happens instead of that?**
Unable to type those keys. \ doesn't behave like the windows key anymore, however the inability to type non-alphanumeric character keys still remains.

maxellnormalbias, 2017-04-20:
I've found a possible workaround: Use Synergy for the mouse and keyboard. However, I can't seem to find any option to disable input capture altogether. `-input none` and `-input /dev/null` are both unfruitful.

irtimmer, 2017-05-27:
Since commit 56f88f4fe8c4e566f59ad7ab70cb51c84599759f the keycodes have been updated for the SDL input stack which you were probably using. So most keys should work correctly now.

closed 2017-05-27 by irtimmer