moonlight-stream/moonlight-embedded #492: Lag/freezing at high bitrates

barhbie, 2017-04-24:
**_NVidia Geforce Experience version:_** GFE 3.5.0.70
**Moonlight Embedded version:** 2.3.0
**Moonlight Embedded running on:** _Raspberry Pi 3

**Output of Moonlight Embedded:**
**_Output start:_**
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Network dropped end of a frame
Network dropped end of a frame
Network dropped end of a frame
Network dropped end of a frame
Network dropped end of a frame
Network dropped end of a frame
Returning RTP packet queued for too long
Received OOS audio data (expected 38236, but got 38237)
Network dropped end of a frame
Waiting for IDR frame
Returning RTP packet queued for too long
Received OOS audio data (expected 38239, but got 38240)
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Network dropped end of a frame
Network dropped end of a frame
Network dropped end of a frame
Network dropped end of a frame
Network dropped end of a frame
Network dropped end of a frame
Waiting for IDR frame
Returning RTP packet queued for too long
Received OOS audio data (expected 38275, but got 38276)
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
[...]

**_Which is followed by this pattern:_**
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Network dropped end of a frame
Network dropped end of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Network dropped end of a frame
Network dropped end of a frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Waiting for IDR frame
Network dropped end of a frame
Network dropped end of a frame
Waiting for IDR frame
Returning RTP packet queued for too long
Received OOS audio data (expected 39167, but got 39168)
[...]

**What is the expected result?**
Streaming at 1080 60fps at a good bitrate.

**What happens instead of that?**
At higher bitrates ( > 10k) I get horrible lag/freezing and the above output from Moonlight.

Moonlight only works properly when I set a bitrate < ~9500. At those bitrates I get a smooth stream, but since the bitrate is so low the picture quality leaves a lot to be desired.

Both the host computer and the Raspberry Pi is hooked up with cat6. I have tried connecting my laptop instead of the Raspberry and running the Moonlight chrome extension, and this works fine at very high bitrates. So the problem doesn't seem to be with my network or the host computer, but rather the combination of RPi + Moonlight-embedded.

My Raspberry is running Raspbian Jessie with Pixel and the latest version of Moonlight-embedded. I've tested the download speed on the unit and gotten a stable ~95 Mb/s.

Despite all this Moonlight-embedded won't perform at bitrates over 10000 kbps.

barhbie, 2017-04-26:
Apparently the problem was the gigabit network speed of my host computer. Switching speed and duplex from "auto negotiation" to 100mbps/Full Duplex on the host solved my problems and I can now stream at 1080p 60fps with near perfect picture quality. 

Thanks for all the work you've put into this amazing software!

closed 2017-04-26 by barhbie

toniprada, 2018-04-09:
This fixed my problem with a new RPI 3B+ and Stretch.