moonlight-stream/moonlight-embedded #493: Can't Pair pi with my pc

lopessiwm, 2017-05-06:
Please provide the following info.

**_NVidia Geforce Experience version:3.5.0.76**
**Moonlight Embedded version: 2.3.0**
**Moonlight Embedded running on:** _Raspberry Pi 3 using WIFI._

**Output of Moonlight Embedded: Can't connect to the server PC **

Can't connect with my PC ip adress on my raspberry pi. Even so to be able to even ping my pc I had to add a new rule on my firewall for it, which i found on the internet and it works, but no way i'm capable of pairing with moonlight.

Is there any solution for it? even if it's my router or something? I'm able to ping from my pc to the pi though.

Thanks for your help

irtimmer, 2017-05-06:
There's nothing we can do if you don't provide the output of Moonlight Embedded.

closed 2017-05-06 by irtimmer

lopessiwm, 2017-05-06:
I just completed the issue on the other one since you closed this one. Is there any chance to open it or something to resolve the problem?