moonlight-stream/moonlight-embedded #494: Can't Pair Pi with PC

lopessiwm, 2017-05-06:
Please provide the following info.

**_NVidia Geforce Experience version:3.5.0.76**
**Moonlight Embedded version:*2.3.0*
**Moonlight Embedded running on:** _Raspberry Pi 3

**Output of Moonlight Embedded:**
sudo moonlight pair 192.168.1.10
Moonlight Embedded 2.3.0 (EMBEDDED;PI)
Connect to 192.168.1.10...
Can't connect to server 192.168.1.10

**What is the expected result?**
I would expect that would moonlight would pair normally with my pc and provide me an pin for my nvdia shield.

**What happens instead of that?**
The moonlight just tries to pair with my pc (and yes i can ping from my pc to the pi and vice versa), althout to ping from my pi to my pc i had to create a rule on my firewall in order to allow that.
the pair attempt just timesout.

irtimmer, 2017-05-06:
Duplicate of #493

closed 2017-05-06 by irtimmer

7hunderbug, 2017-05-06:
Your problem isn't due to an issue with moonlight, it is an issue with 
your network. The Github issue tracker for moonlight isn't really a 
support forum, it is to report bugs. However, in an attempt to be 
helpful I'll suggest the following:

Because you had to create a rule on your firewall simply to do a local 
network ICMP ping, it is likely you will need firewall rules to allow 
all the ports that Nvidia Gamestream requires. A Google search will 
inform you of the ports required to be open. It would be worth checking 
your Windows firewall to make sure the default rules have been added 
when you installed Geforce Experience also. If not, reinstall Geforce 
Experience. You can then determine which ports to open on your network 
firewall comparing the entries in the Windows firewall.