moonlight-stream/moonlight-embedded #495: WeTek Play 2 support

dmnc-net, 2017-05-08:
Hi,

I'm trying to run the moonlight-embedded (but from LibreELEC addon which is outdated (2.2.2/2.2.3)) on the WeTek Play 2 (GPU: Mali MP-450 Penta core). Video is black but I can control the remote PC.

Since this issue is still unresolved, I'd like to ask you/here if it is even possible to support this HW.
Details also here: https://forum.libreelec.tv/thread-6939.html

Can I found some log? When I'm starting it thru ssh, there is nothing relevant.
Thanks.

irtimmer, 2017-05-08:
If I'm correct the WeTek Play 2 does have a Amlogic vpu. While their is code for video output it isn't well supported/maintained as I don't own such a device. To get it to work at least support for it have to be enabled in the distribution provided package and some permissions have to be correctly set. For the last see also https://forum.odroid.com/viewtopic.php?f=91&t=15456&sid=95a03a0d8073ea0b811ed5af01df7566&start=250#p175128

closed 2017-05-08 by irtimmer

gsus24, 2020-04-18:
Hello
i also have blackscreen, but controller works, on wetek play 1 with Amlogic S905H on Android 4.2.2 and moonlight 9.1

Permissions set right, but i have no amstream_hevc. Does it mean, no support for h265 on this box?
In moonlight settings disabled h265, but screen stays black.

@dmnc-net Did you get it working on LibreELEC?

Would be great to play games with the old box.

dmnc-net, 2020-04-27:
@gsus24 no, I didn't managed it. I've switched to the RPi (w/o HEVC) then to regular weak x86 workstation. I think it's not possible or easy to do. Sorry.

gsus24, 2020-04-30:
@dmnc-net This is what i found for Amlogic Devices. Maybe you will give a try
[https://forum.odroid.com/viewtopic.php?t=19974](url)