moonlight-stream/moonlight-embedded #497: Heavy audio distortion after 15~ seconds of streaming

MrSpongen, 2017-05-12:
Please provide the following info.

**_NVidia Geforce Experience version: 3.6.0.74
**Moonlight Embedded version:** 2.4.0
**Moonlight Embedded running on:** Raspberry Pi 2 B+

**Output of Moonlight Embedded:** Stream is working, a bit choppy, but after about 15 seconds, audio goes to full distortion and it's not bearable. Not sure what to answer here...

**What is the expected result?** Audio working.

**What happens instead of that?** Audio, even if I am in the menu of Steam big picture or launch directly into a game, it starts distorting about 15 seconds after stream starts.

irtimmer, 2017-05-18:
Have you tried to run Moonlight Embedded outside of a desktop environment? With the Raspberry Pi 1 and an earlier version of Moonlight Embedded this was a real problem.

MrSpongen, 2017-05-19:
Yes, I tried that as well. It seemed to work slightly better, but the audio was still being distorted after a while. Kept getting the OOS audio data unfortunately. Perhaps my Raspberry Pi 2 B+ can't cope with Moonlight embedded? I got it working like a year ago, but wanted to reformat it and this time this issue kept appearing. I didn't use a wireless dongle back then though. Are there any reports on using one creating a lot of demand out of the Pi or something?

irtimmer, 2017-05-19:
Wireless connections can be very problematic because of packet loss. But that should also result in video corruption. By default Raspbian provides a slow audio decoder (with slightly better sound). So you could try to install http://archive.itimmer.nl/raspbian/libopus0_1.1.2-1_armhf.deb to check if it improves the situation as it would lower the cpu usage. You could also try to use another power adapter as the Raspberry Pi could lower its performance when it hasn't enough power (although I thought you should get a visual indication if this happens)

MrSpongen, 2017-05-19:
My bad. Poor explanation skills showing here. I meant I didn't use a wireless dongle for my Xbox 360 controller, my ethernet cable was plugged in. Sorry for the bad explanation! What I was thinking is that was probably the only "big" change I've done since I tried Limelight as it was known then. Last time I tried streaming, there wasn't this neat embedded solution :)

Also, I got myself one of those specially crafted power adapters for the Raspberry Pi when I got a newer model, when I went from the first to the second. So hopefully my power shouldn't be an issue. 

I'll try the .deb-package.

irtimmer, 2017-06-11:
As there isn't a new reaction I hope the deb package has resolved the issue

closed 2017-06-11 by irtimmer