moonlight-stream/moonlight-embedded #499: Stuttering....Router problem? I have better performance in wifi than ethernet

Dustman71, 2017-05-23:
NVidia Geforce Experience version: 382.05
Moonlight Embedded verion: 2.3.0
Moonlight Embedded running on: Raspberry Pi 3

I do not understand, but I have better performance in wifi than ethernet.
I have tried with my mobile connected in the Wifi (5G) and the result is very smooth but in my Raspberry 3 connected by ethernet I have lag - stuttering.

I have tried different resolutions and fps with the same result.

I assume the problem may be in my Router ???

I hope someone can help me ... I'm desperate ..

regards

7hunderbug, 2017-05-23:
Try turning off IPV6 on your RaspberryPi, you router and your Windows box.

Try forcing your ethernet ports on your router and Windows box to 
100Mbps rather than 1000Mbps / 1Gbps

Dustman71, 2017-05-25:
Too many things that I don't know how to do.
Anyway .. I can't be changing the parameters every time I stream in the Pi
I have only tried to forcing my Windows box to 100Mbps full duplex without results

thank you anyway

irtimmer, 2017-06-11:
If you don't want to change anything there is nothing we can do

closed 2017-06-11 by irtimmer

georgcmef, 2017-06-21:
Kinda running into the same problem. Should I disable ipv6 on the network adapter or should i change the registry key in windows? 

Thank you.