moonlight-stream/moonlight-embedded #504: I can map Xbox 360 pad with xbox.map file

Beric77, 2017-06-06:
Hello,

Im using the last version of Moonlight, running on Raspberrry pi 3 and Raspbian. I just install raspbian and Moonlight, no retropie:

When the system boot, my pad synchronizes with the receptor, and the light indicates it correctly. I try to create the xbox.map on /root/xbox.map and /Pi/xbox.map. Then i try to config buttons (moonlight -map xbox.map) and it finish show this message in both cases:

"Can't open xbox.map"

When i try with sudo moonlight -map xbox.map, it finish without the error message. Then i open the Steam streaming and some buttons do not work. I close the Steam connection and shows this message:

Unmapped button "somenumber..."
Unmapped button "somenumber..."
Unmapped button "somenumber..."
etc.

Other data:

 I have this line on rc.local:
xboxdrv –trigger-as-button –wid 0 –led 2 –deadzone 4000 –dpad-as-button –silent &

Xbox.map content of this post http://embedded-software-architecture.com/?page_id=140

¿What else do I have to configure?

Many thanks.

irtimmer, 2017-06-06:
Do you start moonlight with option `-mapping xbox.map`?

Beric77, 2017-06-06:
Yes, and them i get into Steam and buttons do not work correctly.

Regards

Beric77, 2017-06-09:
Helloooooooooooo ???

irtimmer, 2017-06-11:
The code for mapping files has been rewritten for the new version. So please check if the new version solved your issue.

closed 2017-06-11 by irtimmer