moonlight-stream/moonlight-embedded #505: Raspberry Pi 3 + 1080p60 = big latency

kriegalex, 2017-06-06:
Running moonlight-embedded on 720p30, 720p60 and 1080p30 works very well, even with bitrate at 20000, but as soon as it gets to 1080p60, it goes from perfectly playable to unplayable due to heavy latency (at least 1 second, if not 2).

Is this sudden latency expected or a bug ? I am running moonlight-embedded 2.3.0 on Raspberry Pi 3 with ethernet cable connected directly to PC through one single Netgear switch.

What I find strange is the big jump of latency going from 1080p30 to 1080p60. Is this the limit of the Raspberry Pi 3 CPU hitting hard ? Are there other alternatives to handle 1080p60 other than nvidia shield if this is not a bug?

closed 2017-06-11 by irtimmer

kriegalex, 2017-06-11:
I just looked at the new issue template and don't see what's missing...

irtimmer, 2017-06-11:
The issue template isn't new... and you're missing the output of Moonlight Embedded.