moonlight-stream/moonlight-embedded #507: Several Issues: No Sound on Client, Can't use 1366x768, Big Picture Issues

Crono141, 2017-06-12:
Please provide the following info.

NVidia Geforce Experience version:  3.6.0.74
Moonlight Embedded version: 2.4.0
Moonlight Embedded source: archive.itimmer.nl/raspbian/moonlight
Moonlight Embedded running on: Raspberry Pi 3
Moonlight Embedded running on distribution: RetroPie

**Verbose output `-verbose` of Moonlight Embedded:
moonlight stream 192.168.1.178 -width 1366 -height 768 -bitrate 2000 -verbose
Moonlight Embedded 2.4.0 (EMBEDDED;ALSA;PI)
Connect to 192.168.1.178...
NVIDIA GeForce GTX 970, GFE 3.6.0.74 (gs_03_70_22061309, 7.1.370.0)
Platform Raspberry Pi (Broadcom)
Server doesn't support 1366x768 (60 fps)

moonlight stream 192.168.1.178 -verbose
Moonlight Embedded 2.4.0 (EMBEDDED;ALSA;PI)
Connect to 192.168.1.178...
NVIDIA GeForce GTX 970, GFE 3.6.0.74 (gs_03_70_22061309, 7.1.370.0)
Platform Raspberry Pi (Broadcom)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 12908, but got 12909)
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

What is the expected result?  In the previous version, I was able to set my stream to 1366x768x60 with a bitrate of 2000, and get audio output.  Additionally, the previous version would always show steam big picture even if big picture mode was already running on the host, and it would maintain the hosts BPM state after ending the stream.

What happens instead of that? Now, I get no audio, I cannot stream 1366x768 (even though the resolution exists in the nVidia control panel on the host, and I can switch to it).  If Big PIcture Mode is running on the host when the stream is started, moonlight will kill it on exit and show a cmd window.  If BPM is not running, moonlight will start it, but upon ending the stream, BPM is killed on the host.  IT didn't used to behave this way.  It seemed, at least, that it used to check if BPM was running at the start of the stream, and then return the host to the same state it was in at the end of the stream.  I have noticed this behavior on moonlight for android as well.

Concerning the no audio issue, I noticed that it is muted on the host during the stream, but unmuting it does not solve the issue.  To be clear, there is no audio output on either the client or the host.  If I pass moonlight the -localaudio switch, audio works on the host (like it should with that switch), but I still get no audio on the client.  I have also verified that sound works in other applications on the client, and that audio is NOT muted on the client during the stream.

irtimmer, 2017-06-12:
Could you try to add `-audio hmdi`, `-audio local`, `-audio sysdefault` or `-audio default` to see if any of those options bring audio back?

Crono141, 2017-06-12:
I tried audio alsa but didn't have any luck. My audio out is through the headphone/AV jack. I'll report back in a bit on the other suggestions (can't test right now).

Crono141, 2017-06-13:
Adding -audio default solved the missing audio issue, thanks. I'd still like to troubleshoot the other 2 issues if possible.

irtimmer, 2017-06-13:
What I don't understand is why the previous version had working audio as for a long time HDMI has been the default audio output port, while `-audio local` should also output to the headphone jack.

The BFE issue looks like a bug in Steam or GFE as their are only two commands (launch and cancel) which are send by Moonlight Embedded who can influence application behaviour on the host. Cancel is only run on `moonlight quit` and both implementations haven't changed in a while.

Custom resolutions which aren't marked as supported by GFE are no longer working. Because no other Moonlight clients provided this option, not all GFE versions support it and some users got strange errors when using custom resolutions. In a next version I probably allow the usage of custom resolutions using the `--unsupported` flag.

Crono141, 2017-06-13:
Is there something I could do on the host to work around the issues?  Like perhaps write a script to relaunch BPM after GFE kills it?

irtimmer, 2017-06-19:
You could try to write a script to launch BPM and add it to the GFE gamelist and start is using the `--app` parameter.

The issue with custom resolution has been resolved in the new version and you can now use custom resolutions when adding `--unsupport` as argument.

closed 2017-06-19 by irtimmer

Crono141, 2017-06-22:
Thanks.  I created a shortcut on the host machine to launch BPM.  This works as far as maintaining the state of BPM on the host after streaming.  However, when I quit streaming using the shortcut keys (on a controller: L+R+Select+Start), the host computer remains muted after streaming.

ss-gnalvesteffer, 2021-06-13:
> Could you try to add `-audio hmdi`, `-audio local`, `-audio sysdefault` or `-audio default` to see if any of those options bring audio back?

Not to resurrect this dead issue, but I ran into a similar problem during my first exposure to moonlight, regarding having no sound during the stream. Adding the “-audio sysdefault” option fixed it for me. Hopefully this helps others that land on this issue during their search.

45353, 2024-02-15:
For anyone using rpi 4, you would add `audio = sysdefault` instead of `-audio sysdefault` to the end of the moonlight configuration file at `/opt/retropie/configs/all/moonlight/global.conf`.