moonlight-stream/moonlight-embedded #508: segmentation fault

scaper777, 2017-06-13:
Please provide the following info.

**_NVidia Geforce Experience version: 3.6.0.74
**Moonlight Embedded version: 2.4.0
**Moonlight Embedded source: "deb http://archive.itimmer.nl/raspbian/moonlight jessie main"
**Moonlight Embedded running on: Raspberry Pi
**Moonlight Embedded running on distribution: Retropie

**Verbose output `-verbose` of Moonlight Embedded: i did a strace instead of verbose and came to this

access("hosts/192.168.0.100.conf", R_OK) = -1 ENOENT (No such file or directory)
write(1, "Connect to 192.168.0.100...\n", 28Connect to 192.168.0.100...
) = 28
mkdir("/home", 0775)                    = -1 EEXIST (File exists)
mkdir("/home/pi", 0775)                 = -1 EEXIST (File exists)
mkdir("/home/pi/.cache", 0775)          = -1 EEXIST (File exists)
mkdir("/home/pi/.cache/moonlight", 0775) = -1 EEXIST (File exists)
open("/home/pi/.cache/moonlight/uniqueid.dat", O_RDONLY) = 4
fstat64(4, {st_mode=S_IFREG|0644, st_size=16, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76ef3000
read(4, "d545a81a8fbd15e0", 4096)       = 16
close(4)                                = 0
munmap(0x76ef3000, 4096)                = 0
open("/home/pi/.cache/moonlight/client.pem", O_RDONLY) = -1 ENOENT (No such file or directory)
brk(0x573000)                           = 0x573000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x8} ---
+++ killed by SIGSEGV +++
Segmentation fault

**What is the expected result? to pair my device / map my controls

**What happens instead of that? segmentation fault

irtimmer, 2017-06-13:
It looks like generating a new certificate is going wrong on Debian. Probably caused by the changes needed to be compatible with recent OpenSSL versions. Which has been (incorrectly) reverted for the Raspbian version, but not tested as I currently don't have a Raspberry Pi with Raspbian running.

scaper777, 2017-06-13:
Is it possible to manually create the certificate, and if so, what parameters does it need to have?

irtimmer, 2017-06-13:
You can try to do it manually by looking into mkcert.c but it's easier to just install the previous version from http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/moonlight-embedded_2.3.0-1jessie1_armhf.deb , run `moonlight pair` and reinstall 2.4.0

scaper777, 2017-06-13:
Seems to be fixed now, awesome! Thumbs up for your support

cgutman, 2017-06-14:
I can confirm the crash using the Jessie package. Interestingly it doesn't happen with master when I build locally (using the below simple patch). Maybe 1ed34ff95d26452e029f1cfbea31af473151d7ae didn't make it or something?

`Program received signal SIGSEGV, Segmentation fault.
0x76c1edcc in RSA_generate_key_ex ()
   from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
(gdb) bt
#0  0x76c1edcc in RSA_generate_key_ex ()
   from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
#1  0x76f94bc0 in mkcert ()
   from /usr/lib/arm-linux-gnueabihf/libgamestream.so.1
#2  0x76f94d8c in mkcert_generate ()
   from /usr/lib/arm-linux-gnueabihf/libgamestream.so.1
Cannot access memory at address 0x6a`

`diff --git a/libgamestream/CMakeLists.txt b/libgamestream/CMakeLists.txt
index 22e2bae..8e5c7ac 100644
--- a/libgamestream/CMakeLists.txt
+++ b/libgamestream/CMakeLists.txt
@@ -3,7 +3,7 @@ set(SO_VERSION 1)
 find_package(LibUUID REQUIRED)
 find_package(Threads REQUIRED)
 find_package(CURL REQUIRED)
-find_package(OpenSSL 1.0.2 REQUIRED)
+find_package(OpenSSL 1.0.1 REQUIRED)
 find_package(EXPAT REQUIRED)
 
 pkg_check_modules(AVAHI REQUIRED avahi-client)
diff --git a/libgamestream/client.c b/libgamestream/client.c
index eff1d64..b300a64 100644
--- a/libgamestream/client.c
+++ b/libgamestream/client.c
@@ -493,8 +493,8 @@ int gs_pair(PSERVER_DATA server, char* pin) {
   char client_secret_data[16];
   RAND_bytes(client_secret_data, 16);
 
-  const ASN1_BIT_STRING *asnSignature;
-  X509_get0_signature(&asnSignature, NULL, cert);
+  const ASN1_BIT_STRING *asnSignature = cert->signature;
+  //X509_get0_signature(&asnSignature, NULL, cert);`

irtimmer, 2017-06-14:
I have uploaded a new build to the repository which should fix the issue

closed 2017-06-15 by irtimmer