moonlight-stream/moonlight-embedded #510: Keyboard and Mouse input not being detected.

georgcmef, 2017-06-14:
Hello, just compiled the latest commits and I get no mouse or keyboard input once i connect to my system. 

Running on Pi3
PC is Win10 with GFE 3.7.0.68

here is my output. 

```
Moonlight Embedded 2.4.0 (EMBEDDED;ALSA;PULSE;PI)
Connect to ***************...
NVIDIA GeForce GTX 970 x2, GFE 3.7.0.68 (gs_03_80_22266601, 7.1.380.0)
Platform Raspberry Pi (Broadcom)
No mapping available for /dev/input/event0
No mapping available for /dev/input/event1
No mapping available for /dev/input/event2
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 1950, but got 1951)
Terminated
```

I am guessing that is has to do with the 

```
No mapping available for /dev/input/event0
No mapping available for /dev/input/event1
No mapping available for /dev/input/event2
```

here is what I run on the CL

`moonlight stream ****** -verbose -app MSTC -720`

gabrielpiccolo4, 2017-06-15:
Same here... Also, I don't know if this issue is related, but I'm not being able to map my xbox controller, when I enter the command `moonlight map controller.map` it prints `Connect to controller.map` and then `Can't connect to server controller.map`.

I'm using moonlight 2.4.0.

```
pi@pipplware:~/RetroPie/roms/moonlight $ moonlight map controller.map
Connect to controller.map...
Can't connect to server controller.map
```
```
pi@pipplware:~/RetroPie/roms/moonlight $ ./autostart.sh
Connecting to the PC ...
PC available, starting moonlight
Connect to 192.168.1.103...
No mapping available for /dev/input/event2
No mapping available for /dev/input/event3
No mapping available for /dev/input/event1
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 26583, but got 26584)
No mapping available for /dev/input/event0
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done
```

cgutman, 2017-06-15:
Confirmed that the "No mapping available" warning is the culprit. https://github.com/irtimmer/moonlight-embedded/blob/master/src/input/evdev.c#L422

Keyboard and mouse devices don't require a mapping, so it should not bail out if it cannot find one.

MopheusDG, 2017-06-15:
Same problem here, keyboard and mouse not working. Both rejected because of no mapping.

blu2b, 2017-06-15:
I'm also using a RPi3 and got the same issue!

I tried only connecting a wireless Xbox360 controller. Then a wired Xbox360 controller. Then mouse. Then keyboard.
Don't matter what I do... if I use "-mapping xbox.map" or not.. it's always the same error: "no mapping available[...]"

paincraft0, 2017-06-16:
Same here. PS3 controller not working.
Moonlight Embedded 2.4.0
"No mapping available for /dev/input/event0"

irtimmer, 2017-06-16:
Issue should be solved with commit abbd3766bc33b46abdaffcec4c39cd28b6ef0a07

closed 2017-06-16 by irtimmer

blu2b, 2017-06-17:
I resettet my RPi3 and installed everything new (like described in wiki). Still got the issue trying my mouse or wirless Xbox Controller. 
Am I missing anything?

georgcmef, 2017-06-17:
@blu2b did you do a rebuild of [https://github.com/irtimmer/moonlight-embedded/commit/abbd3766bc33b46abdaffcec4c39cd28b6ef0a07](url) I have yet to do it myself cause I am having trouble compiling Openssl 1.0.2

fadingkite, 2017-06-17:
@georgcmef Pardon my lack of knowledge... Just getting into repositories. What would the command be for rebuilding that specific file?

georgcmef, 2017-06-17:
@fadingkite Not just a single file, you need build the whole program. Don't worry its not to hard. irtimmer has a nice tutorial right here [https://github.com/irtimmer/moonlight-embedded/wiki/Compilation](Wiki). I cant help you past getting the openssl libraries working, I myself am having some problems since the requirements demand a version of openssl that you have to build.

fadingkite, 2017-06-17:
@georgcmef Thanks for the assistance! I'll give that a try next day off.

blu2b, 2017-06-17:
@fadingkite Thanks for the quick help! I'm quite new to all this stuff so my knowledge is very limited. 

I started rebuilding the program one hour ago. (Honestly I thougt there is no need for it when I'm installing everything new)

I used the tutorial you linked but I'm getting errors when I run "cmake ../":

```
The C compiler identification is GNU 4.9.2
Check for working C compiler: /usr/bin/cc
Check for working C compiler: /usr/bin/cc   works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.0.28")
Found Opus: /usr/lib/arm-linux-gnueabihf/libopus.so
Found Broadcom: /opt/vc/include
Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
Could NOT find Amlogic (missing:  AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY)
Found PkgConfig: /usr/bin/pkg-config (found version "0.28")
Checking for module 'libevdev'
Found libevdev, version 1.3
Checking for module 'libudev'
Found libudev, version 215
Checking for module 'sdl2>=2.0.4'

Checking for module 'libavcodec'
No package 'libavcodec' found
Checking for module 'libavutil'
No package 'libavutil' found
Checking for module 'x11'
No package 'x11' found
Checking for module 'vdpau'
No package 'vdpau' found
Checking for module 'libva'
No package 'libva' found
Checking for module 'libva-x11'
No package 'libva-x11' found
Checking for module 'libpulse-simple'
Found libpulse-simple, version 5.0
Checking for module 'libcec>=3.0.0'
 
Checking for module 'egl'
No package 'egl' found
Checking for module 'glesv2'
No package 'glesv2' found
Checking for one of the modules 'uuid'
Looking for pthread.h
Looking for pthread.h - found
Looking for pthread_create
Looking for pthread_create - not found
Looking for pthread_create in pthreads
Looking for pthread_create in pthreads - not found
Looking for pthread_create in pthread
Looking for pthread_create in pthread - found
Found Threads: TRUE
Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.38.0")
CMake Error at /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
system variable OPENSSL_ROOT_DIR: Found unsuitable version "1.0.1t", but
required is at least "1.0.2" (found
/usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so)
Call Stack (most recent call first):
/usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
/usr/share/cmake-3.6/Modules/FindOpenSSL.cmake:370 (find_package_handle_standard_args)
libgamestream/CMakeLists.txt:6 (find_package)

Configuring incomplete, errors occurred!
See also "/home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeError.log".

```

CMakeError.log:
> Determining if the pthread_create exist failed with the following output:
> Change Dir: /home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeTmp
> 
> Run Build Command:"/usr/bin/make" "cmTC_5a542/fast"
> /usr/bin/make -f CMakeFiles/cmTC_5a542.dir/build.make CMakeFiles/cmTC_5a542.dir/build
> make[1]: Entering directory '/home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeTmp'
> Building C object CMakeFiles/cmTC_5a542.dir/CheckSymbolExists.c.o
> /usr/bin/cc     -o CMakeFiles/cmTC_5a542.dir/CheckSymbolExists.c.o   -c /home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
> Linking C executable cmTC_5a542
> /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_5a542.dir/link.txt --verbose=1
> /usr/bin/cc       CMakeFiles/cmTC_5a542.dir/CheckSymbolExists.c.o  -o cmTC_5a542 -rdynamic
> CMakeFiles/cmTC_5a542.dir/CheckSymbolExists.c.o: In function `main':
> CheckSymbolExists.c:(.text+0x38): undefined reference to `pthread_create'
> collect2: error: ld returned 1 exit status
> CMakeFiles/cmTC_5a542.dir/build.make:97: recipe for target 'cmTC_5a542' failed
> make[1]: *** [cmTC_5a542] Error 1
> make[1]: Leaving directory '/home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeTmp'
> Makefile:126: recipe for target 'cmTC_5a542/fast' failed
> make: *** [cmTC_5a542/fast] Error 2
> 
> File /home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
> /* */
> #include <pthread.h>
> 
> int main(int argc, char** argv)
> {
>   (void)argv;
> #ifndef pthread_create
>   return ((int*)(&pthread_create))[argc];
> #else
>   (void)argc;
>   return 0;
> #endif
> }
> 
> Determining if the function pthread_create exists in the pthreads failed with the following output:
> Change Dir: /home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeTmp
> 
> Run Build Command:"/usr/bin/make" "cmTC_c1930/fast"
> /usr/bin/make -f CMakeFiles/cmTC_c1930.dir/build.make CMakeFiles/cmTC_c1930.dir/build
> make[1]: Entering directory '/home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeTmp'
> Building C object CMakeFiles/cmTC_c1930.dir/CheckFunctionExists.c.o
> /usr/bin/cc    -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_c1930.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.6/Modules/CheckFunctionExists.c
> Linking C executable cmTC_c1930
> /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_c1930.dir/link.txt --verbose=1
> /usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_c1930.dir/CheckFunctionExists.c.o  -o cmTC_c1930 -rdynamic -lpthreads
> /usr/bin/ld: cannot find -lpthreads
> collect2: error: ld returned 1 exit status
> CMakeFiles/cmTC_c1930.dir/build.make:97: recipe for target 'cmTC_c1930' failed
> make[1]: *** [cmTC_c1930] Error 1
> make[1]: Leaving directory '/home/pi/Downloads/moonlight-embedded/build/CMakeFiles/CMakeTmp'
> Makefile:126: recipe for target 'cmTC_c1930/fast' failed
> make: *** [cmTC_c1930/fast] Error 2

georgcmef, 2017-06-19:
@blu2b just to make sure that we are working on the same page. Delete the whole moonlight source code folder and download the latest master, do the same for the moonlight-common-c and put that into the "third party" folder. Rebuild and let me know. Cause I was also having openssl problems until i got the latest commit.

blu2b, 2017-06-20:
@georgcmef I directly updated moonlight from the repository and it works. (At least when connecting a mouse. I'm still trying to running a wired/wireless Xbox-Controller) 

Probably it's the easiest way for you just to update on V2.4.1! :)

Salgat, 2017-07-10:
Any update on this? Unfortunately the latest available OpenSSL is 1.0.1 which makes the latest not possible to build from source.

RossWilliamson, 2017-07-23:
Same issue using 2.4.1 so no fix there

TheSpartan94, 2017-08-11:
Any news on this? I still can't compile because of the OpenSSL package required as 1.0.2. 
Could the developers deliver a .deb package for the RPi?

geekycow, 2017-08-11:
Assuming you're building this for the Pi, it doesn't require openssl
1.0.2.  You're probably checking out the master version of the source
rather than the Pi / Raspbian version.  I did the same thing at first.  The
pi/raspbian version of the source is patched for the older openssl.

TheSpartan94, 2017-08-11:
Nope, i have downloaded the files from the raspbian(jessie) branch and it throws this exception:

`
--   Found x11, version 1.6.2
-- Checking for module 'vdpau'
--   No package 'vdpau' found
-- Checking for module 'libva'
--   Found libva, version 0.39.4
-- Checking for module 'libva-x11'
--   Found libva-x11, version 0.39.4
-- Checking for module 'libpulse-simple'
--   Found libpulse-simple, version 5.0
-- Checking for module 'libcec>=3.0.0'
--
-- Checking for module 'egl'
--   No package 'egl' found
-- Checking for module 'glesv2'
--   No package 'glesv2' found
-- Checking for one of the modules 'uuid'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.38.0")
CMake Error at /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR: Found unsuitable version "1.0.1t", but
  required is at least "1.0.2" (found
  /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so)
Call Stack (most recent call first):
  /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.6/Modules/FindOpenSSL.cmake:370 (find_package_handle_standard_args)
  libgamestream/CMakeLists.txt:6 (find_package)
`

Still requiring openssl 1.0.2.