moonlight-stream/moonlight-embedded #511: Moonlight map tries connecting to server

sville2000, 2017-06-15:
I used this guide to set up Moonlight: https://youtu.be/pYoLOoBkEnU

Most recent version of GeForce Experience as of writing
Moonlight 2.4.0
I got Moonlight from the repository
Raspberry Pi 3
RetroPie

I'm trying to map my DualShock 3 controller to Moonlight using
'''
moonlight map PS3.map
"'
which responds with
'''
Connecting to PS3.map...
Can't connect to server PS3.map
'''
When I start up Moonlight, it says that there is no mapping available and it starts Big Picture, but I can't move anywhere. Any advice is helpful, thank you

gabrielpiccolo4, 2017-06-15:
Same here, maybe it's related with this issue: https://github.com/irtimmer/moonlight-embedded/issues/510

```
pi@pipplware:~/RetroPie/roms/moonlight $ moonlight map controller.map
Connect to controller.map...
Can't connect to server controller.map
```

irtimmer, 2017-06-15:
Map commando doesn't exist

closed 2017-06-15 by irtimmer

gabrielpiccolo4, 2017-06-15:
Hey @irtimmer has this command changed? How do we map the controller.map now? Thanks!

jbenleo, 2017-06-16:
Hi @gabrielpiccolo4, I found there is a configuration file /usr/share/moonlight/gamecontrollerdb.txt with all devices ids and configurations. Maybe your controller isn't there. 
I have added mine by copying a similar one (Generic X-Box) at the end of the file and changing the id. You may read the id at file /proc/bus/input/devices ('I' field) and adding four zeros at the end of each parameter.
{Bus}0000{Vendor}0000{Product}0000{Version}0000.
I hope this help you to get your controller working again.