moonlight-stream/moonlight-embedded #512: Client picture freezes after stream start

BlackVodka, 2017-06-17:
Please provide the following info.

**_NVidia Geforce Experience version: ** 3.6.0.74
**Moonlight Embedded versions:**
Raspbian-Repo 2.4.0-2jessie
Github raspbian/jessie (last Commit  Wed Jun 14 20:03:13 2017 +0000)
**Moonlight Embedded sources:**
archive.itimmer.nl/raspbian/moonlight
https://github.com/irtimmer/moonlight-embedded
**Moonlight Embedded running on: **Raspberry Pi 
**Moonlight Embedded running on distribution:** Retropie, most current release

**Verbose output `-verbose` of Moonlight Embedded:**

`pi@RPi3:~ $ moonlight stream -1080 -fps 60 -verbose -app Steam > /tmp/moonlight_steam_verbose.log`
`No mapping available for /dev/input/event0`
`No mapping available for /dev/input/event1`
`No mapping available for /dev/input/event2`
Moonlight Embedded 2.4.0 (EMBEDDED;ALSA;PULSE;PI)
Searching for server...
Connect to 192.168.178.21...
NVIDIA GeForce GTX 1070, GFE 3.6.0.74 (gs_03_70_22061309, 7.1.370.0)
Platform Raspberry Pi (Broadcom)
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Unrecoverable frame 1: 25+2=27 received < 46 needed
Unrecoverable frame 2: 20+1=21 received < 23 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Unrecoverable frame 4: 21+0=21 received < 22 needed
Unrecoverable frame 5: 21+0=21 received < 22 needed
Unrecoverable frame 6: 21+1=22 received < 24 needed
Unrecoverable frame 7: 21+1=22 received < 24 needed
Unrecoverable frame 8: 20+1=21 received < 24 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Unrecoverable frame 11: 20+1=21 received < 23 needed
Unrecoverable frame 12: 20+1=21 received < 24 needed
Unrecoverable frame 13: 20+1=21 received < 24 needed
Unrecoverable frame 14: 20+2=22 received < 24 needed
Unrecoverable frame 15: 21+1=22 received < 24 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Unrecoverable frame 17: 21+0=21 received < 22 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Unrecoverable frame 19: 21+1=22 received < 24 needed
Unrecoverable frame 20: 20+2=22 received < 25 needed
Network dropped an entire frame
Unrecoverable frame 23: 21+0=21 received < 23 needed
Unrecoverable frame 24: 21+0=21 received < 22 needed
Returning RTP packet queued for too long
Received OOS audio data (expected 52403, but got 52404)
Unrecoverable frame 25: 20+0=20 received < 21 needed
Unrecoverable frame 26: 21+2=23 received < 35 needed
Unrecoverable frame 27: 21+1=22 received < 32 needed

[...] (This repeats for quite a while until the stream gets stopped on PC)

Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

(Full output with `-debug`on https://pastebin.com/EZ3Vfw32 )

**What is the expected result?**
Smooth stream in 1080p, 60fps

**What happens instead of that?**
Picture freezes short after Steam starts; sometimes/quite rarely (after some minutes) the pi seems to receive another valid frame and updates the displayed picture.
This looks then like for example https://imghost.io/image/d6Zdk.
Although this isn't the highest quality picture I've ever made you can see the blurred steam UI; this is the 'entry effect'.

**Additional information:**
The Pi is connected with other devices to a FritzBox 7490 router (with internal GBit switch), which is connected through ~ 20 metres of CAT7 ethernet cable (on CAT 6 wall plugs) and a gigabit switch on the other side to my office with my PC in it.
Basically, there's a 1Gbit - connection possible even with the cable to the pi, just the pi slows it down to 100 MBit.

A speed test from windows' ubuntu bash to raspberry pi using nc showed
Server RPi 3 : 
```
nc -vvlnp 12345 > /dev/null
Listening on [0.0.0.0] (family 0, port 12345)
Connection from [192.168.178.21] port 12345 [tcp/*] accepted (family 2, sport 50497)
```
Client Windows PC :
```
dd if=/dev/zero bs=1M count=1K | nc -vvn 192.168.178.22 12345
Connection to 192.168.178.22 12345 port [tcp/*] succeeded!
^C765+0 Datensätze ein
764+0 Datensätze aus
801112064 bytes (801 MB, 764 MiB) copied, 88,3893 s, 9,1 MB/s
```
So, around 70 MBit/s isn't that bad with the pi's internal ethernet interface and shouldn't be a problem AFAIK.
If I understand the output of moonlight correctly, it states transaction failures (`Loss Stats: Transaction failed: 11`) , but why?

Many thanks for help in adance!

irtimmer, 2017-06-19:
Could you try to reduce the network speed of the windows machine to 100mbit to check if it solves your issue?

BlackVodka, 2017-06-20:
Wow, indeed this works, thanks a lot!
As I'm using windows just for gaming and Linux as main operating system, I can totally live with that.

Best regards and thanks for your awesome work!

closed 2017-06-20 by BlackVodka

excelsi0r, 2018-10-21:
Hello, Is there any workaround that does not involve severing down my connection speed? My windows host PC is also my main PC, so, having Gigabit is essential for me.