moonlight-stream/moonlight-embedded #513: PS4 controller not mapping correctly

Kaltern, 2017-06-17:
**_NVidia Geforce Experience version:_3.6.0.74
**Moonlight Embedded version:2.4.0
**Moonlight Embedded source:** Jessie
**Moonlight Embedded running on:** _Raspberry Pi 3
**Moonlight Embedded running on distribution:** RetroPie

**Verbose output `-verbose` of Moonlight Embedded:**

Moonlight Embedded 2.4.0 (EMBEDDED;ALSA;PI)
Connect to 192.168.1.75...
NVIDIA GeForce GTX 1080, GFE 3.6.0.74 (gs_03_70_22061309, 7.1.370.0)
Platform Raspberry Pi (Broadcom)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Error: cannot keep up
Returning RTP packet queued for too long
Received OOS audio data (expected 53369, but got 53370)
Unmapped button: 311
Unmapped button: 311
Unmapped button: 310
Unmapped button: 310
Unmapped button: 310
Unmapped button: 310
Control stream received disconnect event
Input: sendInputPacketOnControlStream() failed: -1
Stopping input stream...done

**What is the expected result?**
PS4 controls correct.

**What happens instead of that?**

Controls are unmapped.  As there is no longer a 'map' command, I expected the built-in mapping to work, instead it appears that the axis' and buttons are just completely incorrect - the gyro for example acts as one of the directions, I don't know which.  Also, the verbose report shows unmapped buttons.

miguas, 2017-06-17:
I have exactly the same issue.
When I try using moonlight map controller.map moonlight tries connecting controller.map server.

Using versing moonlight embedded 2.4.0

pi@retropie:/opt/retropie/configs/moonlight $ moonlight map xbox0.map
Connect to xbox0.map...
Can't connect to server xbox0.map
pi@retropie:/opt/retropie/configs/moonlight $ moonlight map -help
Searching for server...
Connect to 192.168.0.26... ( my local ip address)
map is not a valid action
pi@retropie:/opt/retropie/configs/moonlight $ moonlight map /opt/retropie/configs/moonlight/xbox0.map
Connect to /opt/retropie/configs/moonlight/xbox0.map...
Can't connect to server /opt/retropie/configs/moonlight/xbox0.map

nwildner, 2017-06-18:
Same here with a PS3 controller.

It losts it's old mapping(does not recognizes the mapping file), and whenever i try to map it again, it tries to connect to a "controller.map" server:

irtimmer, 2017-06-19:
`Unmapped button` issue is resolved with commit 0e4820e3a8421c5258272421423baaca8fb98850. Please check if the newly released version solves your issue or provide the new verbose output as some extra debugging information is printed by the new version.

georgcmef, 2017-06-21:
I am also experiencing problems with mapping the DS4 controller. Everything works well except the D-pad. I used sdl Controllermap and it matches whats in gamecontrollerdb.txt. There are actually duplicate entries for my controller ID so I just removed one. SDL reads the dpad as hat while evdev sees each axis as a analog  input. When I connect to my computer dpup results in dpright while dpleft and dpdown do nothing. I have tried changing the number in the mapping and still nothing. 

SDL controllermap output:
`030000004c050000c405000011010000,Sony Computer Entertainment Wireless Controller,platform:Linux,x:b0,a:b1,b:b2,y:b3,back:b8,guide:b12,start:b9,dpleft:h0.8,dpdown:h0.4,dpright:h0.2,dpup:h0.1,leftshoulder:b4,lefttrigger:a3,rightshoulder:b5,righttrigger:a4,leftstick:b10,rightstick:b11,leftx:a0,lefty:a1,rightx:a2,righty:a5,`

irtimmer, 2017-06-21:
The provided gamecontrollerdb.txt is composed from multiple sources which has some overlap. But it shouldn't influence the behavior unless the first is incorrect. It seemed there were still some small bugs in the d-pad code but those have now be resolved in the master branch.

closed 2017-06-21 by irtimmer

georgcmef, 2017-06-21:
Sweet, thank you @irtimmer building now.