moonlight-stream/moonlight-embedded #514: xbox 360 wireless controller isn't mapped

shannonr, 2017-06-19:
Please provide the following info.

**_NVidia Geforce Experience version: GFE 3.6.0.74
**Moonlight Embedded version: 2.4.0
**Moonlight Embedded source: repository
**Moonlight Embedded running on: Raspberry Pi
**Moonlight Embedded running on distribution: Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**
Moonlight Embedded 2.4.0 (EMBEDDED;ALSA;PI)
Connect to 192.168.0.209...
NVIDIA GeForce GTX 1060 6GB, GFE 3.6.0.74 (gs_03_70_22061309, 7.1.370.0)
Platform Raspberry Pi (Broadcom)
No mapping available for /dev/input/event0
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 41482, but got 41483)
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?  The xbox 360 wireless controller should be mapped.

**What happens instead of that? Visualy Moonlight is working correctly but I have no controller mapping so it is unuseable, it doesn't matter if the command is "moonlight stream 192.168.0.209 -verbose -mapping /opt/retropie/configs/moonlight/x360.map" or "moonlight stream 192.168.0.209 -verbose"
 the verbose output is the same and the controller does not work and I have to exit the stream from the host PC.

I also run RetroPi on this Raspberry Pi and the controller works fine in there.

I saw this https://github.com/irtimmer/moonlight-embedded/issues/511 posted the other day and tried it but with no luck.
> > Hi @gabrielpiccolo4, I found there is a configuration file /usr/share/moonlight/gamecontrollerdb.txt with all devices ids and configurations. Maybe your controller isn't there.
> I have added mine by copying a similar one (Generic X-Box) at the end of the file and changing the id. You may read the id at file /proc/bus/input/devices ('I' field) and adding four zeros at the end of each parameter.
> {Bus}0000{Vendor}0000{Product}0000{Version}0000.
> I hope this help you to get your controller working again.

jbenleo, 2017-06-19:
Hi, have you tried to use the new gamecontrollerdb.txt file as the mapping file instead the old .map file?

NiekvanderLinden, 2017-06-19:
Having same issue with latest retropie image and latest moonlight. Adding my Xbox 360 wireless receiver id to the new mapping file doesn't work.

nwildner, 2017-06-19:
It seems that this is a duplicate of https://github.com/irtimmer/moonlight-embedded/issues/513

Any mapping is ignored, and creating a new mapping does not work. It tries to "connect" to an address resolved by the map filename...

irtimmer, 2017-06-19:
Please check if the newly released version solves your issue or provide a new output as some new debugging information is provided by the new version

AutomatedPOS, 2017-06-19:
Don't mean to hijack this thread - but I have the same setup, same issue.
I see from the changelog that "map" was removed.
Updated to the new version - no go.

No mapping available for /dev/input/event2 (030000005e040000a102000000010000)
That ID is missing from gamecontrolerdb.txt.
I'm not familiar with SDL2, so I will tinker a bit.

Hope that helps

shannonr, 2017-06-20:
Will this new version be released on the repository? I am having trouble compiling it.

I've followed the instruction [on the wiki](https://github.com/irtimmer/moonlight-embedded/wiki/Compilation) all the dependencies listed there are installed, cmake seems to want a newer version of openssl but when I apt-get install openssl it tells me it is up to date.

```
pi@retropie:~/moonlight-embedded/build $ cmake ../
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find Amlogic (missing:  AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY)
-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Checking for module 'vdpau'
--   No package 'vdpau' found
-- Checking for module 'libva'
--   No package 'libva' found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
-- Checking for module 'libcec>=3.0.0'
--
-- Checking for module 'egl'
--   No package 'egl' found
-- Checking for module 'glesv2'
--   No package 'glesv2' found
CMake Error at /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR: Found unsuitable version "1.0.1t", but
  required is at least "1.0.2" (found
  /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so)
Call Stack (most recent call first):
  /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.6/Modules/FindOpenSSL.cmake:370 (find_package_handle_standard_args)
  libgamestream/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/moonlight-embedded/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/moonlight-embedded/build/CMakeFiles/CMakeError.log".'''
```

[CMakeOutput.log.txt](https://github.com/irtimmer/moonlight-embedded/files/1086769/CMakeOutput.log.txt)
[CMakeError.log.txt](https://github.com/irtimmer/moonlight-embedded/files/1086768/CMakeError.log.txt)

irtimmer, 2017-06-20:
@AutomatedPOS Could you try to start Moonlight using `SDL_GAMECONTROLLERCONFIG="030000005e040000a102000000010000,X360 Wireless Controller,a:b0,b:b1,back:b6,dpdown:b14,dpleft:b11,dpright:b12,dpup:b13,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3," moonlight` to check if this mapping is correct?

NiekvanderLinden, 2017-06-20:
I'll give it a go with v2.4.1 tonight. Will check the mapping as well.

bgottlob, 2017-06-20:
@irtimmer I was having a similar issue, and that mapping worked perfectly for me on v2.4.1. I had to use the xpad driver instead of following the steps for xboxdrv to get the mapping working.

2000one, 2017-06-20:
Using xboxdrv, all set up according to your Wiki.
Running on boot-up without any option except for --silent

~ $ moonlight stream -verbose -width 1920 -height 1080 -fps 60 -bitrate 20000 -surround XXX.XXX.XXX.XXX
Moonlight Embedded 2.4.1 (EMBEDDED;ALSA;PI)
Connect to XXX.XXX.XXX.XXX...
NVIDIA GeForce GTX 680, GFE 3.7.0.68 (gs_03_80_22266601, 7.1.380.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
No mapping available for /dev/input/event1 (03000000ac0500002102000011010000)
No mapping available for /dev/input/event3 (00000000000000000000000000000000)
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Segmentation fault

Stream starts and steam identifies controller as: Generic X-Input Controller. As soon as a button is pressed stream breaks with error: Segmentation fault

NiekvanderLinden, 2017-06-20:
@2000one there is no mapping for id 03000000ac0500002102000011010000 and 00000000000000000000000000000000 in the [gamecontrollerdb.txt](https://github.com/irtimmer/moonlight-embedded/blob/master/gamecontrollerdb.txt) file.

You could try to copy one there and add it manually to your mappingfile ``` /usr/share/moonlight/gamecontrollerdb.txt ```

2000one, 2017-06-20:
@NiekvanderLinden Thank you, that worked! Now just need to find out how to get or better create correct mapping for my Xbox360 Wireless controller....

NiekvanderLinden, 2017-06-20:
@irtimmer Mapping is partly working, dpad is not working at all, right thumb acts funny (right thumb left always triggering) ect. Also my device id is different ```030000005e040000a102000014010000```.

NiekvanderLinden, 2017-06-20:
In the meantime I downgraded to version ```2.3.0```

```
wget http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/moonlight-embedded_2.3.0-1jessie1_armhf.deb
sudo dpkg --install moonlight-embedded_2.3.0-1jessie1_armhf.deb

# Just to be sure I did a sudo reboot
sudo reboot
```
Then I use this to boot (in file via retropie). 
```
#!/bin/bash
moonlight stream -mapping /opt/retropie/configs/moonlight/controller.map -fps 60 -1080 XXX.XXX.XXX.XXX
```
And with this controller.map file
```
abs_x = 0
abs_y = 1
abs_z = -1
reverse_x = false
reverse_y = true
abs_rx = 3
abs_ry = 4
abs_rz = -1
reverse_rx = false
reverse_ry = true
abs_deadzone = -30792
abs_dpad_x = 16
abs_dpad_y = 17
reverse_dpad_x = false
reverse_dpad_y = false
btn_north = 308
btn_east = 305
btn_south = 304
btn_west = 307
btn_select = 314
btn_start = 315
btn_mode = 316
btn_thumbl = 317
btn_thumbr = 318
btn_tl = 310
btn_tr = 311
btn_tl2 = 312
btn_tr2 = 313
btn_dpad_up = 706
btn_dpad_down = 707
btn_dpad_left = 704
btn_dpad_right = 705
```

I also noticed the xbox controller light was working as expected (show player ect.) Which was not the case when I tried ```2.4.1```. I read something about the difference with ```xpad``` and ```xboxdrv``` that is causing this problem. Also when I was on ```2.4.1``` I could only use 1 controller. 

So my guess is this has something to do with ```xboxdrv``` as well.

2000one, 2017-06-20:
This is my controller (dongle connected directly to Pi3 USB port):

Controller: Microsoft Xbox 360 Wireless Controller (PC)
Vendor/Product: 045e:0719
USB Path: 001:004
Wireless Port: 0
Controller Type: Xbox360 (wireless)

Funny is id 00000000000000000000000000000000 for that controller.
When I use the mapping of 030000005e0400001907000000010000 from the file it works, but everything is messed up really good. I believe that triggers and sticks are switched somehow. Dpad seems to be dead. As soon as I start configuring xboxdrv with usual options like mentioned [here](https://github.com/irtimmer/moonlight-embedded/wiki/Controllers-set-up) the stream will break after input from the right analogue stick with a
`Floating point exception
`
Can't test any more for today...

AutomatedPOS, 2017-06-20:
@irtimmer - Mapping appears to be correct with the command you had me run, to include DPAD.
@shannonr - It was on the repo when I tested it.

2000one, 2017-06-20:
@AutomatedPOS how do you load xboxdrv?
I am using the same mapping as you tested by adding my id to the file and copying the same mapping to the new line, but controls are messed up. Also a floating point execption breaks my stream when I try to load options to xboxdrv and then touch the right stick.

cgutman, 2017-06-21:
Is there a reason you guys are using xboxdrv? It should not be necessary for Xbox 360 or Xbox One gamepads if you're running a 4.9 LTS kernel.

shannonr, 2017-06-21:
I'm not using xboxdrv, I've always just had xpad which comes with Retropie. Also I not sure if it matters but the wireless receiver that is attached my my pi is not a genuine Microsoft receiver but a knock-off from Deal Extreme, that might explain the different ids that are showing up. 

Thanks @AutomatedPOS I got 2.4.1 installed from the repository and with the following added to gamecontrollerdb.txt everything seems to be working for me, I'll do some more testing later today when I have some spare time.

`030000005e040000a102000014010000,X360 Wireless Controller,a:b0,b:b1,back:b6,dpdown:b14,dpleft:b11,dpright:b12,dpup:b13,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Linux,`

Thank you @irtimmer your work is amazing!

sympoz, 2017-06-21:
Using xboxdrv with a PS3 controller over bluetooth, this is the line I had to use, for some reason buttons became all messed up with the new SDL controller config

xboxdrv config, 50% trigger deadzone was required to use analog triggers properly

sudo xboxdrv --evdev /dev/input/event2 --evdev-keymap KEY_#300=y,KEY_#302=a,KEY_#301=b,KEY_#303=x,KEY_#291=start,KEY_#288=back,KEY_#304=guide,KEY_#299=rb,KEY_#298=lb,KEY_#292=du,KEY_#295=dl,KEY_#294=dd,KEY_#293=dr,KEY_#289=tl,KEY_#290=tr, --evdev-absmap ABS_#13=rt,ABS_#12=lt,ABS_#0=x1,ABS_#1=y1,ABS_#2=x2,ABS_#3=y2 --axismap -Y1=Y1,-Y2=Y2 --mimic-xpad --deadzone-trigger 50% --detach-kernel-driver --deadzone 3000 --dpad-as-button & sleep 1

030000005e0400008e02000010010000,Microsoft X-Box 360 pad,a:b4,b:b5,back:b10,dpdown:b1,dpleft:b2,dpright:b3,dpup:b0,guide:b12,leftshoulder:b8,leftstick:b13,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b9,rightstick:b14,righttrigger:a5,rightx:a3,righty:a4,start:b11,x:b6,y:b7,

To map your controller in gamecontrollerdb.txt with the new method:

1. Find your SDL controller id (its a calculated guid only used by SDL)
  *launch moonlight and look for "no mapping available for xxxxxxxxxxxxxxxx", you can find name with evtest but ive not found the name essential. you can obviously see how it formatted in /usr/share/moonlight/gamecontrollerdb.txt

2. in a terminal launch "jstest /dev/input/jsX" where X is the joystick id you want to use, eg js0 or js1 if your using xboxdrv as js0 is used by xboxdrv to create js1.

3. use jstest to find the button you want to map. I press the button I want moonlight to use for A and jstest tells me it is button 4. eg for me, moonlight looks to gamecontrollerdb.txt and maps my /dev/input/event3 button 4 to the moonlight gamepad button A, left stick x and y to axis 0 and 1 etc.

4. do this for all the buttons you want mapped, you can see I am using axis 2 and 5 for left and right trigger, put your new line into gamecontrollerdb.txt.

I hope this helps somebody.

This process is a little more simple if you dont use xboxdrv, provided your controller has an event id. Unfortunately xboxdrv is the only way I have found to get analog triggers working for racing games as sixad does not support deadzone or sensitivity config. I also tried setting the deadzone within steam but it was not enough.

For debugging I emptied gamecontrollerdb.txt and used just the line I wanted just to save any confusion. You can map as many controllers as you want with this method and moonlight will always look to the same mapping file. Thanks for the great work @irtimmer

shannonr, 2017-06-21:
Had a chance to try `2.4.1` out and everything is working perfectly, Thanks for all the help.

2000one, 2017-06-21:
@cgutman I use xboxdrv because it is recommended everywhere in set up guides and in the wiki. I had a good working installation back in 11/2016 with raspbian and xboxdrv which I had not used for a while now and started fresh a few days ago. The according controller entry seems to have been removed now/today....

next tests with xpad:

- I set up a fresh retropie installation today and all the available updates (former experiences were under standard raspbian installations with xpad disabled and xboxdrv installed like in the now missing wiki section)

- configured the game controller with the graphical UI
- exited emulation station and started the stream through ssh connection
- my controller ID is now 030000005e040000a102000000010000 the same ID as @AutomatedPOS has got, I use the original MS Wireless Receiver
- added ID to gamecontrollerdb 
- copied the mapping of X360 Wireless to my new ID line 

experience in steam big picture mode:

- RS = not working at all
- RS click = unmapped button 318
- DPAD = could not observe any reaction in steam
- LS = up+down not working
- LS click = could not observe any reaction in steam
- A = working
- B = working
- X = unmapped button 307
- Y = unmapped button 308
- START = unmapped button 315
- GUIDE = working
- BACK = could not observe any reaction in steam
- LB = working
- RB = unmapped button 311
- RT = could not observe any reaction in steam
- LT = could not observe any reaction in steam

~ $ moonlight stream -verbose -1080 -fps 60 -bitrate 20000 XXX.XXX.XXX.XXX

Moonlight Embedded 2.4.1 (EMBEDDED;ALSA;PI)
Connect to XXX.XXX.XXX.XXX…
NVIDIA GeForce GTX 680, GFE 3.7.0.68 (gs_03_80_22266601, 7.1.380.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected X360 Wireless Controller (030000005e040000a102000000010000) on /dev/input/event0
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 24932, but got 24933)
Unmapped button: 318
Unmapped button: 311
Unmapped button: 307
Unmapped button: 308
Unmapped button: 315

irtimmer, 2017-06-21:
Main reason to switch to the SDL gamecontrollerdb.txt format is to be able to make gamepads work out-of-the-box. Unless someone use a custom controller configuration using xboxdrv. Beside [SDL2](https://www.libsdl.org/), the format is also used by [Gnome Games](https://wiki.gnome.org/Design/Playground/Games), [Godot Engine](https://godotengine.org/). And therefore mappings from them [1](https://github.com/emscripten-ports/SDL2/blob/master/src/joystick/SDL_gamecontrollerdb.h), [2](https://github.com/gabomdq/SDL_GameControllerDB/blob/master/gamecontrollerdb.txt) and [3](https://github.com/godotengine/godot/blob/master/main/input_default.cpp) can easily be reused.

For the creation of mappings there are multiple options, like [Steam Big Picture](https://steamcommunity.com/app/225260/discussions/0/846945955316594293), [SDL2 Gamepad Tool](http://www.generalarcade.com/gamepadtool/) and SDL2 own [Controllermap](https://github.com/spurious/SDL-mirror/blob/master/test/controllermap.c), although you have to run these on a Linux system as a different OS or driver result in different mappings.

A good alternative for jstest is [sdl2-jstest](https://github.com/Grumbel/sdl-jstest) as is uses the SDL2 gamepad input and therefore doesn't depend on /dev/input/js*.

There are currently some issues left in the Moonlight Embedded implementation which can result in a floating point exception or segmentation fault when using incorrect mappings, random input when gamepad has additional unmapped axis or buttons. And beside the included mappings, SDL2 also has some fallbacks for XBox controllers which are not (yet) included in Moonlight Embedded.

2000one, 2017-06-21:
I validated the correct mapping of my controller with jstest /dev/input/js0
I also found my ID 030000005e040000a102000000010000 in your link to [godot](https://github.com/godotengine/godot/blob/master/main/input_default.cpp) with exactly the same mapping config as I am using now.

Still the controls are messy as described [above](https://github.com/irtimmer/moonlight-embedded/issues/514#issuecomment-310029601).

Do I have to mess with Steam settings on host now to get the controller working?
Where can I start looking for issues?

Thanks for any hints!

irtimmer, 2017-06-21:
@2000one could you run `evtest /dev/input/event0` and provide the list of Supported events? Hopefully this explains why there are still some unmapped buttons.

2000one, 2017-06-21:
@irtimmer testing the inputs shows corresponding event logs...
```
~ $ evtest /dev/input/event0
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x45e product 0x2a1 version 0x100
Input device name: "Xbox 360 Wireless Receiver"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
    Event code 704 (BTN_TRIGGER_HAPPY1)
    Event code 705 (BTN_TRIGGER_HAPPY2)
    Event code 706 (BTN_TRIGGER_HAPPY3)
    Event code 707 (BTN_TRIGGER_HAPPY4)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   -512
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 1 (ABS_Y)
      Value   2113
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 2 (ABS_Z)
      Value      0
      Min        0
      Max      255
    Event code 3 (ABS_RX)
      Value   2411
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 4 (ABS_RY)
      Value   -266
      Min   -32768
      Max    32767
      Fuzz      16
      Flat     128
    Event code 5 (ABS_RZ)
      Value      0
      Min        0
      Max      255
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 96 (FF_GAIN)
Properties:
Testing ... (interrupt to exit)
```

2000one, 2017-06-21:
Here is a table with the outputs from my controller. I made it to make sure everything registered right according to the mapping. Looks right to me...
maybe this overview can be useful to anybody else.

### Xbox 360 Wireless Receiver

| evtest /dev/input/event* event code | evtest button name | SDL_GAMECONTROLLERCONFIG button name | jstest /dev/input/js* button/axis number |
|-------------------------------------|--------------------|--------------------------------------|------------------------------------------|
| 304                                 | BTN_SOUTH          | a                                    | b0                                       |
| 305                                 | BTN_EAST           | b                                    | b1                                       |
| 307                                 | BTN_NORTH          | x                                    | b2                                       |
| 308                                 | BTN_WEST           | y                                    | b3                                       |
| 310                                 | BTN_TL             | leftshoulder                         | b4                                       |
| 311                                 | BTN_TR             | rightshoulder                        | b5                                       |
| 314                                 | BTN_SELECT         | back                                 | b6                                       |
| 315                                 | BTN_START          | start                                | b7                                       |
| 316                                 | BTN_MODE           | guide                                | b8                                       |
| 317                                 | BTN_THUMBL         | leftstick                            | b9                                       |
| 318                                 | BTN_THUMBR         | rightstick                           | b10                                      |
| 704                                 | BTN_TRIGGER_HAPPY1 | dpleft                               | b11                                      |
| 705                                 | BTN_TRIGGER_HAPPY2 | dpright                              | b12                                      |
| 706                                 | BTN_TRIGGER_HAPPY3 | dpup                                 | b13                                      |
| 707                                 | BTN_TRIGGER_HAPPY4 | dpdown                               | b14                                      |
| 0                                   | ABS_X              | leftx                                | a0                                       |
| 1                                   | ABS_Y              | lefty                                | a1                                       |
| 2                                   | ABS_Z              | lefttrigger                          | a2                                       |
| 3                                   | ABS_RX             | rightx                               | a3                                       |
| 4                                   | ABS_RY             | righty                               | a4                                       |
| 5                                   | ABS_RZ             | righttrigger                         | a5                                       |
| 16                                  | ABS_HAT0X          | dpleft/dpright                       | b11/b12                                  |
| 17                                  | ABS_HAT0Y          | dpdown/dpup                          | b14/b13                                  |

sympoz, 2017-06-22:
@2000one make sure there is not an existing mapping with the same guid, that happened to me and I found an existing guid mapping in the gamecontrollerdb.txt, that is why I emptied the file and only had 1 entry

2000one, 2017-06-22:
@sympoz Thanks! I can confirm, that the ID of my controller 030000005e040000a102000000010000 is not already listed in the mapping file.

What I now realized is, that during copying and pasting in nano yesterday I made a really stupid mistake.
My mapping line was incomplete which I did not doublecheck afterwards in the too small terminal window I used.
That is why some buttons where missing. Sorry for the confusion!!

...But...

the DPAD is still not working when I use the [gamecontrollerdb.txt](https://github.com/irtimmer/moonlight-embedded/blob/master/gamecontrollerdb.txt) file for mapping and just add my mapping to it.
Everything works perfectly when I use the manual command with mapping as suggested [here](https://github.com/irtimmer/moonlight-embedded/issues/514#issuecomment-309678582) or just have the mapping as the only one in the file.

In detail:

everything is **working with commandline** mapping

`SDL_GAMECONTROLLERCONFIG="030000005e040000a102000000010000,X360 Wireless Controller,a:b0,b:b1,back:b6,dpdown:b14,dpleft:b11,dpright:b12,dpup:b13,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Linux," moonlight stream -verbose -1080 -fps 60 -bitrate 20000 XXX.XXX.XXX.XXX`

I get a **non working** DPAD **with** following **line added** to gamecontrollerdb.txt, or with ID changed in front of already existing mapping.

`030000005e040000a102000000010000,X360 Wireless Controller,a:b0,b:b1,back:b6,dpdown:b14,dpleft:b11,dpright:b12,dpup:b13,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Linux,`

but **everything is working with above only as the only single line** in gamecontrollerdb.txt

or am I too stupid again?

sympoz, 2017-06-24:
@2000one sounds like a formatting problem maybe? definately strange, but if an empty file works with the only line, just use that unless you have other controllers you need mapping for

2000one, 2017-06-24:
@sympoz yes, I also tried to just replace the controller ID in the already existing line of the whole database file to avoid any formatting errors. Does not work. Funny.

alfalfa1978, 2017-06-24:
@2000one Thank you! After deleting all other controllers an adding just one line with the right ID of my X360 wireless controller, my dpad is also working now.
It works also with two controllers.

groschn, 2017-06-24:
Hello everyone,

First of all Thank you for your great work @irtimmer . the Wohle stremaing thing works perfectly. Sadly i am experiencing the same issues with my Xbox 360 Wireless Controller. It has the same ID as @2000one but it does not werk with his config. It seems that the D-Pad is not working. I tried the commandline aswell as putting the config in the designated Configfile but none of that worked. 

Has anyone another Idea how this can be fixed?

2000one, 2017-06-26:
@groschn have you also already tried to empty the gamecontrollerdb.txt and add your mapping as the only single line to the file?

The mapping I use is copied from the already existing mapping from the database, so should theoretically work for you.

You can check the events provided by your controller as described [here](https://github.com/irtimmer/moonlight-embedded/issues/514#issuecomment-310060651) and test the output with jstest like described [here](https://github.com/irtimmer/moonlight-embedded/issues/514#issuecomment-309963932) under point 2.
You may check my [table](https://github.com/irtimmer/moonlight-embedded/issues/514#issuecomment-310163270) as reference as well.

hshar7, 2017-06-30:
Please also add 030000005e040000a102000007010000 to the list.. you have to support a fallback for xbox360 controllers at least to solve this common problem.

2000one, 2017-06-30:
@hshar7 does your DPAD function when you add your controller ID and mapping to the database file?

CaptChair, 2017-07-01:
Hey all!

Figured i'd join in to this too.

Currently got an xbox one controller finally not crashing on first button hit. Copied and pasted an existing xbox one configuration from the controllerdb.txt and put my controller ID in front of it.

030000005e040000d102000003020000,Microsoft X-Box One USB,x:b2,a:b0,b:b1,y:b3,back:b6,guide:b8,start:b7,dpleft:h0.8,dpdown:h0.0,dpdown:h0.4,dpright:h0.0,dpright:h0.2,dpup:h0.0,dpup:h0.1,leftshoulder:h0.0,leftshoulder:b4,lefttrigger:a2,rightshoulder:b5,righttrigger:a5,leftstick:b9,rightstick:b10,leftx:a0,lefty:a1,rightx:a3,righty:a4,platform:Linux,

None of the dpad works except up. up is treated as right.

here's what I have from jstest

b0 - a
b1 - b
b2 - x
b3 - y
b4 - left bumper
b5 - right bumper
b6 - left trigger
b7 - right trigger
b8 - select
b9 start
b10 - xbox
b11 - left thumb click
b12 - right thumb click

axes 0 left stick left and right movement
axes 1 left stick up and down movement
axes 2 right stick left and right movement
axes 3 right stick up and down movement
axes 4 dpad left and right movement
axes 5 dpad up and down movement

I see a lot of mention of a for axis, and a few mentions of hat.up in my google searches. hoping someone can point me in the right direction as everything else seems to work. 

what do these actually represent?
h0.0 
h0.1 
h0.2 
h0.4 
h0.8
I'm assuming it's hat up, down, left, right, but since mine shows as axes in the test, am I barking up the wrong tree? also curious that none of the other dpad buttons except up do anything.

What should my next steps be? been at this for about 2 hours now.

2000one, 2017-07-01:
@CaptChair
please see my posts. Having DPAD issues with 360 Wireless Controller. Works only with commandline mapping or as the only mapping in database file.

Did you try emptying your db file and adding your mapping as the only single line in the file?

CaptChair, 2017-07-01:
@2000one

Thanks for the quick reply. I've gone through the entire post and completed all recommended steps, including command line mapping, and removing all other controllers from the db :(

I know i shouldn't need the xboxdrv drivers now, but should I maybe try getting those?

CaptChair, 2017-07-02:
Update: With my DS4 and the USB adapter, it does the same thing with the DPAD, even when it's the only one in the list as well.

irtimmer, 2017-07-02:
D-Pad issue is fixed with commit dd23c70d46f8e924e6bc3ca0411fa4f998021bfc

closed 2017-07-02 by irtimmer

CaptChair, 2017-07-02:
So I'm really new to all of this but eager to learn and try to figure stuff out. Do I just need to run an update to get the dpad fix, or do I have to do something manually?

deelowe, 2017-08-20:
I dont think this has been fixed. I'm still having issues with the xbox 360 wireless dpad mapping. Right is down, Left is up.

Same setup as 2000one:

- guid: 030000005e040000a102000000010000
- name: "Xbox 360 Wireless Receiver"

Using retropie. Driver is xpad.

1031982, 2017-11-14:
deelowe, using your configuration Moonlight sees my controller, however it gives an error saying that it's not configured correctly.
Any ideas?

tomismrkolj, 2017-12-12:
Hi,
I have another issue. I am working on mapping Microsoft sidewinder wheel to moonlight.
All buttons are mapped OK, as well as left/right steering. Accerlerator and brake pedals are mapped to right and left triggers, but in resting state (not pressed) they are giving full throttle and full brakes. If I press them, then the car can be driven, albight with strange feeling of depressing the pedals instead of pressing them...

So question: how to invert axis in gamecontrollerdb.txt?

Tomi