moonlight-stream/moonlight-embedded #515: 17240 Floating point exception

jbenleo, 2017-06-19:
Please provide the following info.

**_NVidia Geforce Experience version:_3.6.0.74
**Moonlight Embedded version: 2.4.0
**Moonlight Embedded source:http://archive.itimmer.nl/raspbian/moonlight jessie main
**Moonlight Embedded running on: Rasbperry Pi 3
**Moonlight Embedded running on distribution: OSMC

**Verbose output `-verbose` of Moonlight Embedded:
Moonlight Embedded 2.4.0 (EMBEDDED;ALSA;PI)
Connect to 192.168.2.1...
NVIDIA GeForce GTX 1060 6GB, GFE 3.6.0.74 (gs_03_70_22061309, 7.1.370.0)
Platform Raspberry Pi (Broadcom)
Stream 1280 x 720, 30 fps, 5000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 57972, but got 57973)
[ERROR] USBController::on_write_data(): USB write failure: 12: LIBUSB_TRANSFER_ERROR
17240 Floating point exception

**What is the expected result?
command: moonlight stream -verbose -720 -fps 30 -mapping /home/osmc/Steam/gamecontrollerdb.txt
Play Shadow of Mordor. Failure happens after loading current game.

**What happens instead of that?
moolight exits with the specified error:
17240 Floating point exception

irtimmer, 2017-06-19:
USBController::on_write_data() is an error from xboxdrv and not from Moonlight Embedded

jbenleo, 2017-06-20:
I've seen the floating point exception without the xboxdrv error and even without starting any game (at steam big picture).
It happens when i use the right stick of my controller so it seems to be a mapping problem.
I will go further this evening to see what's happening.

2000one, 2017-06-20:
Just noticed that I am having a similar problem like mentioned [here](https://github.com/irtimmer/moonlight-embedded/issues/514#issuecomment-309853009) in issue #514 

Is moonlight breaking streams when xboxdrv is messing up?

jbenleo, 2017-06-23:
Finally I could test with xpad instead xboxdrv and everything is ok but the dpad configuration.
Currently I am having the same issue than @2000one at #514 about the mapping in gamecontrollerdb.txt or in command line.

@irtimmer If it could help, i could configure the xboxdrv again and run moonlight with valgrind or another debugging tool in order to provide some extra information of the floating point exception.

2000one, 2017-06-24:
@jbenleo do you mean that mapping also only works for you through command line or as the only mapping in the gamecontrollerdatabase?

irtimmer, 2017-07-02:
Version 2.4.2 should resolve the issues with crashing.

closed 2017-07-02 by irtimmer