moonlight-stream/moonlight-embedded #517: Audio desyncs w/ video after few minutes

neoesquire, 2017-06-21:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.7.0.68
**Moonlight Embedded version:** 2.4.1
**Moonlight Embedded source:** repo
**Moonlight Embedded running on:** Raspberry Pi 3 B
**Moonlight Embedded running on distribution:** Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**
Moonlight Embedded 2.4.1 (EMBEDDED;ALSA;PI)
Connect to 192.168.1.122...
NVIDIA GeForce GTX 970, GFE 3.7.0.68 (gs_03_80_22266601, 7.1.380.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 47312, but got 47313)
Detected X360 Wireless Controller (030000005e040000a102000000010000) on /dev/inp                                                                                        ut/event2
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
Audio in sync with video for duration of the stream.

**What happens instead of that?**
Audio desyncs after a few minutes and the gap between the audio and video grow as the stream is active.

**What happens instead of that?**
Audio desyncs after a few minutes and the gap between the audio and video grow as the stream is active.

I've tried other streaming solutions on the same hardware and they kept their sync for the entire time they were streaming. I don't experience any audio or video jutters, and the desync happens even if just left on the Steam dashboard.

irtimmer, 2017-06-25:
Duplicate of #280

closed 2017-06-25 by irtimmer

benwilesmusic, 2017-09-29:
I'm not sure this is a duplicate; #280 seems to refer to an audio lag that sometimes happens when starting moonlight, while this one refers to a sync problem whereby the audio drifts: while the audio is in sync at the start of the stream, it gets further out of sync as the stream continues. 

I have observed this latter problem.