moonlight-stream/moonlight-embedded #519: Xbox One controller DPad does not work properly in Moonlight latest

jasonridesabike, 2017-06-26:
Using Geforce Experience latest and Moonlight from the irtimmer repo.

DPad (hat switch in sdl2) doesn't work. Used sdl2-jstest and as far as I can tell the mapping in gamecontrollerdb.txt are correct, but no dice.

Tested using Xbox One S controller and Xbox One original controllers. One S controller's ID isn't in the gamecontrollerdb.txt, but copying the line for Xbox One orig controller and changing the ID fixes that.

2000one, 2017-06-27:
please see #514, specifically [here](https://github.com/irtimmer/moonlight-embedded/issues/514#issuecomment-310371314) and following.

Boarti, 2017-07-01:
Hello,

I had the same issue with the wonky dpad, and mine works now in bigpicture.

i used the moonlight stream -verbose -1080 -fps 60 -Steam command, which told me the ID of the controller.

Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected Microsoft X-Box One pad v2 (030000005e040000dd02000003020000) on /dev/input/event1

in the meantime i connected the controller to my windows pc and started sdl2 gamepadtool.
it recognized my controller right away, and i clicked on copy mapping string, which displays the map, i copied it ( a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b10,leftshoulder:b4,leftstick:b8,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b9,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3," and edited the existing entry in the gamecontrollerdb.txt on the raspberry.

wich yielded this entry:
030000005e040000dd02000003020000,Microsoft X-Box One pad v2,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b10,leftshoulder:b4,leftstick:b8,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b9,righttrighttri$ghttgger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Linux,

qyark, 2017-07-01:
I'm having a similar issue with my ps4 controller. I have the controller mapped using the gamecontrollerdb, and I don't recieve any errors when starting my stream, but the dpad is completely unresponsive. All other bottons/sticks works fine. This is the line from my map
050000004c050000c405000000010000,Sony Computer Entertainment Wireless Controller,platform:Linux,x:b0,a:b1,b:b2,y:b3,back:b8,guide:b12,start:b9,dpleft:h0.8,dpdown:h0.4,dpright:h0.2,dpup:h0.1,leftshoulder:b4,lefttrigger:a3,rightshoulder:b5,righttrigger:a4,leftstick:b10,rightstick:b11,leftx:a0,lefty:a1,rightx:a2,righty:a5,

irtimmer, 2017-07-02:
D-Pad issue is fixed in commit dd23c70d46f8e924e6bc3ca0411fa4f998021bfc

closed 2017-07-02 by irtimmer

qyark, 2017-07-02:
How do I apply this update?

cmcpasserby, 2017-07-02:
@qyark currently you would have to pull and compile your own, but i would assume that it will eventually end up in in the itimmers repo and be update about via apt-get

cmcpasserby, 2017-07-03:
@irtimmer do you intend to update your packages for Raspbian/OSMC, i am just noticing those are still at 2.4.0 as opposed to the 2.4.2 packages you got for ArchLinux?