moonlight-stream/moonlight-embedded #520: Moonlight stops after hitting a button

pausanjim11, 2017-06-27:
Please provide the following info.

**_NVidia Geforce Experience version:_ 3.7.0.81**
**Moonlight Embedded version: 2.4.1**
**Moonlight Embedded source:**deb http://archive.itimmer.nl/raspbian/moonlight jessie main
**Moonlight Embedded running on:** _Raspberry Pi 3 (RetroPie)..._
**Moonlight Embedded running on distribution:** _Retropie..._

**Verbose output `-verbose` of Moonlight Embedded:**
moonlight -verbose
Moonlight Embedded 2.4.1
Usage: moonlight [action] (options) [host]
       moonlight [configfile]

 Actions

        pair                    Pair device with computer
        unpair                  Unpair device with computer
        stream                  Stream computer to device
        list                    List available games and applications
        quit                    Quit the application or game being streamed
        help                    Show this help

 Global Options

        -config <config>        Load configuration file
        -save <config>          Save configuration file
        -verbose                Enable verbose output
        -debug                  Enable verbose and debug output

 Streaming options

        -720                    Use 1280x720 resolution [default]
        -1080                   Use 1920x1080 resolution
        -4k                     Use 3840x2160 resolution
        -width <width>          Horizontal resolution (default 1280)
        -height <height>        Vertical resolution (default 720)
        -fps <fps>              Specify the fps to use (default -1)
        -bitrate <bitrate>      Specify the bitrate in Kbps
        -packetsize <size>      Specify the maximum packetsize in bytes
        -codec <codec>          Select used codec: auto/h264/h265 (default auto)
        -remote                 Enable remote optimizations
        -app <app>              Name of app to stream
        -nosops                 Don't allow GFE to modify game settings
        -localaudio             Play audio locally
        -surround               Stream 5.1 surround sound (requires GFE 2.7)
        -keydir <directory>     Load encryption keys from directory
        -mapping <file>         Use <file> as gamepad mappings configuration file
        -platform <system>      Specify system used for audio, video and input: pi/imx/aml/x11/x11_vdpau/sdl/fake (default auto)
        -unsupported            Try streaming if GFE version or options are unsupported

 I/O options (Not for SDL)

        -input <device>         Use <device> as input. Can be used multiple times
        -audio <device>         Use <device> as audio output device

Use Ctrl+Alt+Shift+Q or Play+Back+LeftShoulder+RightShoulder to exit streaming session

**What is the expected result?**

Controller works fine

**What happens instead of that?**

I hit a button and the stream stops by giving an error like that:

./Steam.sh: línea 2: 17640 Violación de segmento  moonlight stream 192.168.1.23 -1080

hshar7, 2017-06-30:
Having same issue.. segmentation fault when I hit any button as soon as stream starts.
What happened to your map option?
I cannot do $ moonlight map controller.map

plugpoint, 2017-06-30:
I fixed this by removing the mapping option from my .sh file.

In 2.4.0 release notes it says this: 
"Use SDL2 style gamepad mappings and therefore map function is dropped"

CaptChair, 2017-07-01:
Those having the crashes:

I had the same issue, and it was because my Xbox One controller's id didn't exist in the gamecontrollers.txt file

look for an error like this: No mapping available for /dev/input/event0 (030000005e040000d102000003020000)

Grab the number (030000005e040000d102000003020000 in my case)

then 

sudo nano /usr/share/moonlight/gamecontrollerdb.txt

look for a controller that may be another "revision" of yours, copy it, and paste it at the end of the gamecontrollerdb.txt file

replace number at the start of the line with the number from your warning

In my case

030000005e040000dd02000003020000,Microsoft X-Box One pad v2,x:b2,a:b0,b:b1,y:b3,back:b6,guide:b8,start:b7,dpleft:h0.8,dpdown:h0.0,dpdown:h0.4,dpright:h0.0,dpright:h0.2,dpup:h0.0,dpup:h0.1,leftshoulder:h0.0,leftshoulder:b4,lefttrigger:a2,rightshoulder:b5,righttrigger:a5,leftstick:b9,rightstick:b10,leftx:a0,lefty:a1,rightx:a3,righty:a4,platform:Linux,

became

030000005e040000d102000003020000,Microsoft X-Box One USB,x:b2,a:b0,b:b1,y:b3,back:b6,guide:b8,start:b7,dpleft:h0.8,dpdown:h0.0,dpdown:h0.4,dpright:h0.0,dpright:h0.2,dpup:h0.0,dpup:h0.1,leftshoulder:h0.0,leftshoulder:b4,lefttrigger:a2,rightshoulder:b5,righttrigger:a5,leftstick:b9,rightstick:b10,leftx:a0,lefty:a1,rightx:a3,righty:a4,platform:Linux,

And VOILA! My crashes were gone!

Now the dpad is all sorts of not working for me currently (up moves right, other directions do nothing which is actually why i'm here. But I'm hoping if I can help some people get passed the crashes, we can get through dpad issues together :))

Quaked, 2017-07-01:
Same bug here...

irtimmer, 2017-07-02:
Version 2.4.2 should resolve the issues with crashing.

closed 2017-07-02 by irtimmer