moonlight-stream/moonlight-embedded #521: Controller not mapping.

penkin, 2017-06-28:
**_NVidia Geforce Experience version:_** 3.7.0.81
**Moonlight Embedded version:** 2.4..1
**Moonlight Embedded source:** deb http://archive.itimmer.nl/raspbian/moonlight jessie main
**Moonlight Embedded running on:** _Raspberry Pi_
**Moonlight Embedded running on distribution:** _Retropie_

**Verbose output `-verbose` of Moonlight Embedded:**

Moonlight Embedded 2.4.1
Usage: moonlight [action] (options) [host]
       moonlight [configfile]

 Actions

	pair			Pair device with computer
	unpair			Unpair device with computer
	stream			Stream computer to device
	list			List available games and applications
	quit			Quit the application or game being streamed
	help			Show this help

 Global Options

	-config <config>	Load configuration file
	-save <config>		Save configuration file
	-verbose		Enable verbose output
	-debug			Enable verbose and debug output

 Streaming options

	-720			Use 1280x720 resolution [default]
	-1080			Use 1920x1080 resolution
	-4k			Use 3840x2160 resolution
	-width <width>		Horizontal resolution (default 1280)
	-height <height>	Vertical resolution (default 720)
	-fps <fps>		Specify the fps to use (default -1)
	-bitrate <bitrate>	Specify the bitrate in Kbps
	-packetsize <size>	Specify the maximum packetsize in bytes
	-codec <codec>		Select used codec: auto/h264/h265 (default auto)
	-remote			Enable remote optimizations
	-app <app>		Name of app to stream
	-nosops			Don't allow GFE to modify game settings
	-localaudio		Play audio locally
	-surround		Stream 5.1 surround sound (requires GFE 2.7)
	-keydir <directory>	Load encryption keys from directory
	-mapping <file>		Use <file> as gamepad mappings configuration file
	-platform <system>	Specify system used for audio, video and input: pi/imx/aml/x11/x11_vdpau/sdl/fake (default auto)
	-unsupported		Try streaming if GFE version or options are unsupported

 I/O options (Not for SDL)

	-input <device>		Use <device> as input. Can be used multiple times
	-audio <device>		Use <device> as audio output device

Use Ctrl+Alt+Shift+Q or Play+Back+LeftShoulder+RightShoulder to exit streaming session

**What is the expected result?**
Steam to not shutdown after I hit a button on my Xbox 360 Wireless controller.

**What happens instead of that?**
Steam starts up as expected but soon as I hit any button it shuts down.

**Additional Info**
I'm unable to create a controller map using `moonlight map xbox.map`. It tries to connect to server `xbox.map` which is obviously incorrect. Not sure if I'm missing something to create the controller map.

I tried getting a map online and created it manually but running `moonlight stream -720 -30fps 10.0.0.3 -app Steam -input /dev/input/event0 -mapping /opt/retropie/configs/moonlight/xbox.map` I get the following...

```
moonlight: unrecognized option '-30fps'
Connect to 10.0.0.3...
No mapping available for /dev/input/event0 (030000005e040000a102000000010000)
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 6948, but got 6949)
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done
```

Seems to not find a mapping for my controller `No mapping available for /dev/input/event0 (030000005e040000a102000000010000)` and when I click a button it quits out of Steam.

Running `evtest` I can see the controller is connected and at the correct input that I specified.

```
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	Xbox 360 Wireless Receiver
```

Any help would be greatly appreciated.

penkin, 2017-06-28:
Fixed this issue. All I had to do is add `030000005e040000a102000000010000` to the `gamecontrolerdb.txt` file. Now I just need to locate the correct bindings for the Xbox 360 Wireless controller.

D3laru3, 2017-07-03:
Hi penkin, pls where did you put this string? i am having the same issue...I put it instead of xwc string on 2nd line

dmnc-net, 2017-07-03:
@penkin please any details about your solution? I've no `gamecontrolerdb.txt` file on my retripie system

D3laru3, 2017-07-03:
@dmnc-net its in /usr/share/moonlight directory

penkin, 2017-07-05:
@D3laru3 sorry, did not see the comments. I simply put `030000005e040000a102000000010000` in the bottom of `gamecontrolerdb.txt` file and copied the strings on one of the other xbox wireless controllers. This did not work as the botton mapping is all messed up. Have not had any time to fix that yet. Have you come right?

D3laru3, 2017-07-05:
@penkin No I`ve had no time as well. But I will try it ... I`ve tried only my approach but error with no mapping available still persist...I`ve found a lot of mappings so i will try it soon... I`ll stay in touch...So basically if your approach will work, we will need only working mapping string...

penkin, 2017-07-06:
@D3laru3 here is what I have got working so far...

In `/usr/share/moonlight/gamecontrollerdb.txt` I have this;

`030000005e040000a102000000010000,X360 Wireless Controller,a:b0,b:b1,back:b6,dpdown:b14,dpleft:b11,dpright:b12,dpup:b13,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Linux,`

That is the only line I have in the entire file. I removed all the other controller definitions as one of the posts suggested that maybe the other definitions were messing it up. I also tried with the original file by putting that line first, last and everywhere. Still did not help.

The controller works but the keys are just all mapped wrong.

Remember that `030000005e040000a102000000010000` is the controller's identification. My controller is plugged into the Raspberry Pi.

Hope that helps getting this sorted.

D3laru3, 2017-07-06:
Have you tried to put Xbox 360 Wireless Receiver instead of X360 Wireless Controller string in gamecontrollerdb.txt? 
I am just guessing. cause error is about Xbox 360 Wireless Receiver. At least mine. But i think that this is only cosmetical value. I do not know how do they parse this file.
Have you tried how signals and keys are respresented? For example: a:b0 is maybe a:0.
I think that there is some command to check these keys. 
It is "evtest" or "evtest /dev/input/event0"

penkin, 2017-07-06:
Thanks @D3laru3. I tried the `evtest` and this was the output of the (Y) button:

```
Event: time 1499364465.065678, type 1 (EV_KEY), code 308 (BTN_WEST), value 1
Event: time 1499364465.065678, -------------- EV_SYN ------------
Event: time 1499364465.176695, type 1 (EV_KEY), code 308 (BTN_WEST), value 0
Event: time 1499364465.176695, -------------- EV_SYN ------------
```

Now how does that relate back to the `/usr/share/moonlight/gamecontrollerdb.txt` file?

D3laru3, 2017-07-06:
ok, so it looks like that code for the Y btn is 308. value is the state (clicked and released). so you could try to change y:b3 to y:308 or y:b308 in gamecontrollerdb.txt file.

D3laru3, 2017-07-06:
it will not work i think. I checked the implementation and it is not working as I thought. I will check it deeply tomorrow. Maybe i will find something.

CaptChair, 2017-07-06:
So honest question. Have you tried writing down what buttons are doing what, and manually modifying the mapping yourself?

I mean if x does it's job, find the binding for x and shift it to y, do the same for all other keys.

penkin, 2017-07-07:
@CaptChair I have not done that for two reasons, time and gosh that seems tedious. Maybe when I get some free time I can go down that path.

irtimmer, 2017-07-07:
The mapping isn't very difficult. When you start evtest you will get a list of supported events. The first button maps to b0, the second to b1 etc... The same holds for the abs and hats, although they are mixed in the evtest output. Therefore its easier to use own of the available mapping tools as gamepadtool (see also https://github.com/irtimmer/moonlight-embedded/issues/514#issuecomment-310039899). The reason the provided mappings are incorrect seems to be caused by the use of RetroPie (see also https://github.com/irtimmer/moonlight-embedded/issues/528#issuecomment-313638232)

closed 2017-07-07 by irtimmer

penkin, 2017-07-07:
@D3laru3 here is the string I generated using http://generalarcade.com/gamepadtool/

`030000005e040000a102000000010000,XBox Wireless Controller,a:b0,b:b1,x:b2,y:b3,back:b6,start:b7,leftstick:b8,rightstick:b9,leftshoulder:b4,rightshoulder:b5,dpup:h0.1,dpdown:h0.0,dpleft:h0.8,dpright:h0.2,leftx:a0,lefty:a1,rightx:a3,righty:a4,lefttrigger:a2,righttrigger:a5,platform:Windows,`

Will give it a test and let you know how it goes.

irtimmer, 2017-07-07:
@penkin the mapping probably won't work as its generated for Windows and doesn't take the RetroPie driver changes into account.

penkin, 2017-07-07:
@irtimmer gosh that sucks. No idea what to do then in that case. Just did a quick test and I see it did not work. I don't have access to a Linux box. So seems I cannot use RetroPie and Moonlight together then I guess?

irtimmer, 2017-07-07:
I'm not aware of any easy method to create a mapping from RetroPie. So you have to create a mapping manually. 

Maybe you can try the mapping from #528 which is a changed mapping for the Xbox 360 controller on RetroPie
`030000005e040000a102000000010000,X360 Controller,a:b0,b:b1,back:b8,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b10,leftshoulder:b4,leftstick:b11,lefttrigger:b6,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b12,righttrigger:b7,rightx:a2,righty:a3,start:b9,x:b2,y:b3,platform:Linux,`

RossWilliamson, 2017-07-23:
Having no luck here getting XBOX wireless gamepad to work.  Using v....  Have removed all entries in gamecontrolerdb.  It now only has the following:

030000005e040000a102000007010000,Xbox 360 Wireless Receiver (XBOX),a:b0,b:b1,back:b6,dpdown:b14,dpleft:b11,dpright:b12,dpup:b13,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Linux,

When running moonlight stream I get:

No mapping available for Xbox 360 Wireless Receiver (XBOX) (030000005e040000a102000007010000) on /dev/input/event3

geekycow, 2017-07-23:
I think the latest patches in the master will fix this, but as a workaround
for now try replacing `030000005e040000a102000007010000` with `default`
instead.  You may still get that error message but in my tests the mappings
were still applied.

DocMAX, 2017-08-14:
same here. i get:

Add input /dev/input/event20...
Detected Xbox 360 Wireless Receiver (030000005e040000a102000000010000) on /dev/input/event20 as Xbox 360 Wireless Receiver
No mapping available for Xbox 360 Wireless Receiver (030000005e040000a102000000010000) on /dev/input/event20

using "default" as workaround

cg-david, 2017-11-23:
Hey everyone. 
I had the same problem. 
I am using RetroPie where you have to configure your Controller anyway. From there I also new which button is associated with which button ID. It was easy, as the template of `/usr/share/moonlight/gamecontrollerdb.txt` used the same button IDs as in the RetroPie setup: 

- `b[ID]`  always means ID of button that was pushed, 
- while  `a[ID]` refers to the joysticks. 
- there's a third setup for the D-Pad, but I don't understand it, but I also don't need to :)

My Xbox One Controller had the ID `030000005e040000d102000001010000`
I also played around with `jstest /dev/input/js[ID]` with ID being 0 for my first joystick.

It finally brought me to the following setup in `/usr/share/moonlight/gamecontrollerdb.txt` : 

`030000005e040000d102000001010000,Microsoft X-Box One pad,platform:Linux,x:b2,a:b0,b:b1,y:b3,back:b8,guide:b10,start:b9,dpleft:h0.8,dpdown:h0.0,dpdown:h0.4,dpright:h0.0,dpright:h0.2,dpup:h0.0,dpup:h0.1,leftshoulder:b4,lefttrigger:b6,rightshoulder:b5,righttrigger:b7,leftstick:b11,rightstick:b12,leftx:a0,lefty:a1,rightx:a3,righty:a4,`

When starting moonlight, it will complain about wrong button mapping, but as of now, I haven't faced any issues so far while playing Pro Evolution Soccer. 

One thing I'd like to comment on. When reading comments like:

> I have not done that for two reasons, _time_ and gosh that seems _tedious_

> Just did a _quick_ test

I'd suggest buying something like Steam Link or so. Moonlight on Raspberry and RetroPie is not (yet) an official package and IMHO can be considered as development. I really appreciate the work of irtimmer and everyone working on solutions to improve that. I guess Coding / problem solving requires some fiddling and experimenting. After all, the package RetroPie contains is tremendous and can not be paid by money but only by dedication and time!

emilekberg, 2017-12-30:
@cg-david this mapping does not seem to be fully correct. it's better, but the right analogue stick is mapping incorrectly (at least when playing geometry wars) the ship would only fire hard left or right, not 360 degrees as it should.

danieldabate, 2018-01-21:
Hi @emilekberg, try this config:
`030000005e040000d102000001010000,Microsoft X-Box One pad,platform:Linux,x:b2,a:b0,b:b1,y:b3,back:b8,guide:b10,start:b9,dpleft:h0.8,dpdown:h0.0,dpdown:h0.4,dpright:h0.0,dpright:h0.2,dpup:h0.0,dpup:h0.1,leftshoulder:b4,lefttrigger:b6,rightshoulder:b5,righttrigger:b7,leftstick:b11,rightstick:b12,leftx:a0,lefty:a1,rightx:a2,righty:a3,`

It worked for me flawlessly. 

Thanks @cg-david for your config, it was very helpful.

NinjaChurch, 2018-01-21:
Thanks @danieldabate 
That config worked for my Xbox One controller.

cg-david, 2018-01-23:
@emilekberg 
Already a month since I posted this, but I had some problems afterwards too. There is a really easy command line program, showing you the button you are pushing atm. The pattern is quite easy, I just can't remember the command to use. If @danieldabate hasn't solved this yet, PM me or answer to the thread. I'll try to help. 
Thanks @danieldabate for the edit and @NinjaChurch for the confirmation.

juicypop, 2018-01-31:
@cg-david
Can you help me to map my xbox one wireless controller? I've spend the last few days getting it connected, mapping it in emulationstation just to find out that my moonlight isn't working any more.

I tried evtest and jstest but can't figure out what how I have to map the buttons from neither one :(

cg-david, 2018-02-01:
Hi juicypop. 
evtest and jstest are usually the way to go, as they give you the output of your joystick buttons. 
Are they working in Emulationstation? Have you tried it in games?

juicypop, 2018-02-01:
The controller works as intended in ES 
But as soon as I start moonlight the controller isn't responding though it is list in the controllerdb 

I just don't know if it's a messed up configuration in the db or if something else is causing the problem that my controller won't work in steam big picture 

I also don't know how to read and convert the outputs of evtest and jstest to manually edit the mappings

cg-david, 2018-02-21:
Hey juicypop. 
Okay, so the issue is within moonlight. I think it's the configuration. 
Well in jstest you have the axes and the buttons. And buttons correlate with b0 to b[x]. They should change in jstest if you push a button like A. If you see, that pressing A changes something in button 3, then A correlates to b3. At least that's how I derived the mapping.

cg-david, 2018-02-21:
Oh and two further notes: 
1) There are multiple Xbox One Mappings in the text-file of moonlight. Make sure you change the right one ;) 
2) It takes some trial and error. Don't give up if the problem doesn't solve within 5mins.

stevoisiak, 2018-09-09:
Why is this issue labeled `wont-fix`?