moonlight-stream/moonlight-embedded #522: General questions

Henster523, 2017-07-02:
Hi
I'm new to GitHub and raspberry pi's in general. I have a few questions and didn't know where to ask them. Is this the right place? Anyway I was wondering what controllers work with moonlight for the raspberry pi? Also I'm wanting to play 2 player coop games like Gang beasts, Portal 2 And L4D2 among others. Is it possible to use 2 controller on one raspberry pi?

closed 2017-07-02 by irtimmer

CaptChair, 2017-07-03:
Basically, any controller for the most part has some form of functionality. If it's not in the db, you can manually add it yourself in most cases.

As for 2 player, not sure tbh.

AgentCrowe, 2017-07-06:
@Henster523 I haven't found a Steam game that was designed for 2 players that hasn't worked the way I wanted it to for 2 players on my RP3, I enjoy 2 player Rocket League and Gungeon the most. I run 2 Wii U Pro controllers and I only had to set the first one up with Moonlight through the controller.map thing After that the other controller (Which obviously needs to be synced in the Pi) works flawlessly without any additional work.

Henster523, 2017-07-07:
Thanks for the replies.