moonlight-stream/moonlight-embedded #526: Pressing right stick of Xbox 360 wireless controller leads to steam menu

peroanjo, 2017-07-03:
GeForce Experience version: 
**Latest**
Moonlight Embedded version:
**2.4.2**
**Moonlight Embedded source:** 
repository
**Moonlight Embedded running on:** 
Raspberry
**Moonlight Embedded running on distribution:** 
OSMC

**What is the expected result?**
When playing a game launched through Steam Big Picture mode, expected a response to pressing the right stick. I run moonlight through the Luna Launcher for Kodi.

**What happens instead of that?**
Instead, steam menu opens as if I was pressing the xbox central button.

PD: I hope I'm posting this correctly. Sorry if I cause any inconvenience, as you can see I'm a noob. Also, sorry for any grammatical mistake.

CaptChair, 2017-07-04:
You may need to find your controller in the gamecontrollerdb.txt and edit the bindings. I would also double check my steam controller settings to ensure that it's not just wrongly set in there

irtimmer, 2017-07-04:
I just fixed a bug for the XBox mappings (76cf6f633949c35a6e4d99c201b45f990c411400), but currently you have to compile from source to test is. So I hope someone with the same issue can check if this solves the issue.

peroanjo, 2017-07-05:
So I launched some games directly from Geforce Experience to make sure steam wasn't messing my controller and i realized that pressing any of the sticks just doesn't do anything. Maybe it isn't mapped?

geekycow, 2017-07-20:
I had the same issue with an Xbox 360 Wireless Controller.  It's a default vs wireless xbox 360 controller mapping difference; it's using the default configuration you need to be using xwc.

As a workaround for now, I copied the `xwc` configuration from gamecontrollerdb.txt into a new file and changed `xwc` to `default` and the button mappings were then correct, but if I'm reading it correctly, Iwan's above referenced bugfix will have it working in the next version as it should automatically detect and use the xbox 360 wireless controller mapping by default in future (assuming you are also using such a controller and that it's being named as such by the xpad kernel driver.)

(When the new pi binary is in the repo I'll make a point of mentioning here if this has fixed the issue for me.)

geekycow, 2017-07-20:
Hi,

TL;DR Yes, it works perfectly.  :)

I undid my workaround and then I tried compiling the raspbian-jessie version with the two recent commits from the master, including [https://github.com/irtimmer/moonlight-embedded/commit/76cf6f633949c35a6e4d99c201b45f990c411400](url) and that fixes the issue(s) entirely on my end (Raspberry Pi 2 + Raspbian + a generic eBay xbox 360 wireless controller receiver + a genuine Microsoft XBox 360 Wireless Controller.)  Controller properly detected and properly mapped by default now.  Fantastic!  (Admittedly, it took me many hours to work out how to do it, with accidents like thinking I had to install the latest openssl, which I eventually managed, and then downgraded again when I realised I didn't need to do it for the raspbian version, but at least I learnt something and contributed in some way... and I've got a fully working setup now!  Yay.  Thank you for your hard work Iwan. :) )

Regards
Julie
x

Gudui, 2017-08-05:
I've made a fix for Debian Jessie which solves the issue.
I think you have to uninstall moonlight-embedded first, and then execute:
sudo dpkg -i ./moonlight-embedded_2.4.2-1jessie1_armhf.deb

to install it w
[moonlight-embedded_2.4.2-1jessie1_armhf.zip](https://github.com/irtimmer/moonlight-embedded/files/1202478/moonlight-embedded_2.4.2-1jessie1_armhf.zip)
ith the latest two commits.

closed 2017-08-06 by irtimmer

stevoisiak, 2018-09-24:
I’m seeing the same issue occur on a wired Xbox One gamepad.

While in Steam’s web browser:

- Right stick up = Move cursor left
- Right stick right = Open Steam menu
- Right stick down = Move cursor right
- Right stick left = Open Steam menu