moonlight-stream/moonlight-embedded #529: EVTEST works, Jtest Works, but unable to Map controller

Dublino, 2017-07-07:
**_NVidia Geforce Experience version:3.7.0.81**
**Moonlight Embedded version:2.4.1**
**Moonlight Embedded source:** Jessie Main..._
**Moonlight Embedded running on:** _Raspberry Pi 3..._
**Moonlight Embedded running on distribution:** Retropie/..._
Using an Xbox 360 Controller connected through a ZettaGuard receiver, and using xpad driver. 
I have also tried on a Hama (generic) USB controller. Controllers works on Retropie and when testing via jtest /dev/input/js0 

and evtest /dev/input/event0
pi@retropie:/opt/retropie/configs/moonlight $ evtest /dev/input/event0
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x45e product 0x2a1 version 0x107
Input device name: "Xbox 360 Wireless Receiver (XBOX)"
Supported events:

**Verbose output `-verbose` of Moonlight Embedded:**

pi@retropie:/opt/retropie/configs/moonlight $ moonlight -verbose map controller.map
Moonlight Embedded 2.4.1 (EMBEDDED;ALSA;PI)
Connect to controller.map...
Can't connect to server controller.map

I tried Debug as well 
pi@retropie:/opt/retropie/configs/moonlight $ moonlight -debug map controller.map
Moonlight Embedded 2.4.1 (EMBEDDED;ALSA;PI)
Connect to controller.map...
Request https://controller.map:47984/serverinfo?uniqueid=2eb9f89cbb908d43&uuid=1baa8b58-c779-4c79-bf8b-6ca317952a67
 Same results 

**What is the expected result?**
When Mapping Controller, I would expect to be prompted for input from controller 

**What happens instead of that?**
Connecting to controller service times out 

Is Mapping no longer supported?

JoePata, 2017-07-07:
From my understanding the "map" command no longer exists. `/usr/share/moonlight/gamecontrollerdb.txt` is used instead for mapping purposes.

If you run `moonlight stream -verbose` you should see ID of controller you are using and find it in the file mentioned above. E.g. `Detected Microsoft X-Box 360 pad (030000005e0400008e02000014010000) on /dev/input/event0 as X360 Controller`.

If you can't find that ID in file you can add another entry with the ID and mapping (can base on other xbox mapping most likely). 

I had to edit that file for mapping of xbox 360 wired controller to work properly. Used `jstest /dev/input/js0` to help determine correct mapping. You can view the update I had to make for controller to work in issue #528 I just opened.

Hope that helps. Cheers!

irtimmer, 2017-07-07:
Duplicate of #511

closed 2017-07-07 by irtimmer