moonlight-stream/moonlight-embedded #532: Orange Pi dependencies compile error

Maor545, 2017-07-12:
**Moonlight Embedded version: 2.4.2
**Moonlight Embedded source: from source
**Moonlight Embedded running on: Orange Pi One
**Moonlight Embedded running on distribution: armbian - ubuntu xenial desktop with legacy kernel 5.30

**Verbose output `-verbose` of Moonlight Embedded: attached in log

**What is the expected result? to compile and run

**What happens instead of that?
I am trying to compile on Orange Pi One but getting issues. I have armbian 5.30 and installed following dependencies. I am not able to compile. I would be grateful for any  help.

answers in #249 do not help much.
[moonlight.log.txt](https://github.com/irtimmer/moonlight-embedded/files/1140555/moonlight.log.txt)

Maor545, 2017-07-22:
after painfull instalation of the libraries, which were not included in the wiki page I am stuck with compilation.

I got to building with error:

> [ 61%] Building C object CMakeFiles/moonlight.dir/src/video/ffmpeg_vaapi.c.o
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:23:33: fatal error: libavutil/hwcontext.h: No such file or directory
compilation terminated.
CMakeFiles/moonlight.dir/build.make:422: recipe for target 'CMakeFiles/moonlight.dir/src/video/ffmpeg_vaapi.c.o' failed
make[2]: *** [CMakeFiles/moonlight.dir/src/video/ffmpeg_vaapi.c.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/moonlight.dir/all' failed
make[1]: *** [CMakeFiles/moonlight.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Maor545, 2017-07-23:
I made an empty hwcontext.h and hwcontext_vaapi.h and got a few new errors:

> root@orangepione:~/moonlight-embedded/build# make
[ 38%] Built target moonlight-common
[ 59%] Built target gamestream
[ 61%] Building C object CMakeFiles/moonlight.dir/src/video/ffmpeg_vaapi.c.o
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c: In function ‘va_get_format’:
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:32:25: warning: implicit declaration of function ‘av_hwframe_ctx_alloc’ [-Wimplicit-function-declaration]
   AVBufferRef* hw_ctx = av_hwframe_ctx_alloc(device_ref);
                         ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:32:25: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:38:3: error: unknown type name ‘AVHWFramesContext’
   AVHWFramesContext* fr_ctx = (AVHWFramesContext*) hw_ctx->data;
   ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:38:32: error: ‘AVHWFramesContext’ undeclared (first use in this function)
   AVHWFramesContext* fr_ctx = (AVHWFramesContext*) hw_ctx->data;
                                ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:38:32: note: each undeclared identifier is reported only once for each function it appears in
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:38:50: error: expected expression before ‘)’ token
   AVHWFramesContext* fr_ctx = (AVHWFramesContext*) hw_ctx->data;
                                                  ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:39:9: error: request for member ‘format’ in something not a structure or union
   fr_ctx->format = AV_PIX_FMT_VAAPI;
         ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:40:9: error: request for member ‘sw_format’ in something not a structure or union
   fr_ctx->sw_format = AV_PIX_FMT_NV12;
         ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:41:9: error: request for member ‘width’ in something not a structure or union
   fr_ctx->width = context->coded_width;
         ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:42:9: error: request for member ‘height’ in something not a structure or union
   fr_ctx->height = context->coded_height;
         ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:43:9: error: request for member ‘initial_pool_size’ in something not a structure or union
   fr_ctx->initial_pool_size = MAX_SURFACES + 1;
         ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:45:7: warning: implicit declaration of function ‘av_hwframe_ctx_init’ [-Wimplicit-function-declaration]
   if (av_hwframe_ctx_init(hw_ctx) < 0) {
       ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:51:10: error: ‘AVCodecContext {aka struct AVCodecContext}’ has no member named ‘hw_device_ctx’
   context->hw_device_ctx = device_ref;
          ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:52:10: error: ‘AVCodecContext {aka struct AVCodecContext}’ has no member named ‘hw_frames_ctx’
   context->hw_frames_ctx = hw_ctx;
          ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c: In function ‘va_get_buffer’:
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:58:10: warning: implicit declaration of function ‘av_hwframe_get_buffer’ [-Wimplicit-function-declaration]
   return av_hwframe_get_buffer(context->hw_frames_ctx, frame, 0);
          ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:58:39: error: ‘AVCodecContext {aka struct AVCodecContext}’ has no member named ‘hw_frames_ctx’
   return av_hwframe_get_buffer(context->hw_frames_ctx, frame, 0);
                                       ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c: In function ‘vaapi_init_lib’:
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:62:10: warning: implicit declaration of function ‘av_hwdevice_ctx_create’ [-Wimplicit-function-declaration]
   return av_hwdevice_ctx_create(&device_ref, AV_HWDEVICE_TYPE_VAAPI, ":0", NULL
          ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:62:46: error: ‘AV_HWDEVICE_TYPE_VAAPI’ undeclared (first use in this function)
   return av_hwdevice_ctx_create(&device_ref, AV_HWDEVICE_TYPE_VAAPI, ":0", NULL
                                              ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c: In function ‘vaapi_queue’:
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:72:3: error: unknown type name ‘AVHWDeviceContext’
   AVHWDeviceContext* device = (AVHWDeviceContext*) device_ref->data;
   ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:72:32: error: ‘AVHWDeviceContext’ undeclared (first use in this function)
   AVHWDeviceContext* device = (AVHWDeviceContext*) device_ref->data;
                                ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:72:50: error: expected expression before ‘)’ token
   AVHWDeviceContext* device = (AVHWDeviceContext*) device_ref->data;
                                                  ^
/root/moonlight-embedded/src/video/ffmpeg_vaapi.c:73:3: error: unknown type name ‘AVVAAPIDeviceContext’
[... 56 lines cut ...]
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Multimedia Maintainers <pkg-multimedia-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 941 kB
Depends: libavutil-ffmpeg54 (= 7:2.8.11-0ubuntu0.16.04.1)
Conflicts: libavutil-libav-dev
Homepage: https://ffmpeg.org/
Supported: 5y
Download-Size: 269 kB
APT-Manual-Installed: yes
[... cut ...]
 cutting edge.
 .
 This library is a utility library to aid portable multimedia programming.
 It contains safe portable string functions, random number generators, data
 structures, additional mathematics functions, cryptography and multimedia
 related functionality (like enumerations for pixel and sample formats).
 It is not a library for code needed by both libavcodec and libavformat.
 The goal for this library is to be modular, small, efficient and useful.
 .
 This package contains the development files.

7hunderbug, 2017-07-23:
You are missing the hwcontext.h ffmpeg/libav development header file. This is obvious from the compile output. You either need to install the ffmpeg or libav dev packages for your distribution or you need to compile them from scratch.

This isn't a bug in moonlight so it is inappropriate to file an issue here on github. The github issue tracker isn't a support forum. I don't mean to be rude (I have offered help to you above regardless).
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

Maor545, 2017-07-23:
i will try to compile the libs. If there is a working support forum, I would move my questions there. On the other hand, the moonlight wiki page is not much helpfull and does not list all dependencies, so arter I get it working, I will offer myself to contribute to the wiki.

7hunderbug, 2017-07-23:
A quick check of the contents of libbavutil-dev on packages.ubuntu.com shows that the version in Ubuntu xenial is too old to contain the hwcontext header files. You will either need to compile libav from scratch or find a backported deb package. The version released for Ubuntu Zeisty does have these files. Ubuntu Yakkety also does not.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

adriancuzman, 2017-07-26:
To compile moonlight embedded on Orange PI with Armbian 5.30 you need to compile ffmpeg yourself. 
These are the dependencies that I needed to make it work, besides the dependencies from wiki:

sudo apt-get install vdpau-driver-all
sudo apt-get install libvdpau-dev

--- compile ffmpeg (I  used version 3.3.2)
sudo apt-get install pkg-config
wget http://ffmpeg.org/releases/ffmpeg-x.x.x.tar.bz2
tar xjvf ffmpeg-x.x.x.tar.bz2
cd ffmpeg-x.x.x
./configure --enable-shared 
make
sudo make install
--- end

--missing dependencies for moonlight specific to OrangePi
sudo apt-get install libegl1-mesa-dev
sudo apt-get install libmali-sunxi-dev
sudo apt-get install libenet-dev

After is compiled it works with vdpau acceleration, even 1080p at 60fps with no lag :)
The only issue is that it hangs on exit... only powering off and on will bring it back to life...

adriancuzman, 2017-07-26:
also, there are missing #ifdefs when VAAPI acceleration is missing, function calls are made to VAAPI even if it's not available, so compilation will fail, I've made a pull request with the modifications, you can get the modification before the pull request is merged. Just a heads up, after all the dependencies  are fixed and ffmpeg is compiled.

Maor545, 2017-07-27:
Thx, I am trying to compile ffmpeg, but Orange Pi One has too little RAM to get it done. It will take time to setup crosscompile.

irtimmer, 2017-08-06:
VAAPI issues have been solved by merging pull request #537

closed 2017-08-06 by irtimmer

endlessmind, 2017-08-06:
@adriancuzman 
I've tried to build moonlight after your suggestion on Debian running on OrangePi Lite.
ffmpeg and vdpau compiled fine, but running cmake on moonlight-embedded source says no video available.
vdpauinfo works fine and typing "ffmpeg" gives me version name and stuff like that.
I've attached logfile.
I've no idea where to go from here.
[logfile.txt](https://github.com/irtimmer/moonlight-embedded/files/1203451/logfile.txt)

Maor545, 2017-08-13:
@endlessmind 
from your log I say, install the dependencies, I suppose you forgot to do "make install" after compiling ffmpeg, because you are still missing dependencies.