moonlight-stream/moonlight-embedded #534: quit option doesn't work

posti85, 2017-07-21:
**NVidia Geforce Experience version:** 3.7.0.81
**Moonlight Embedded version:** 2.4.2-1jessie1
**Moonlight Embedded source:** repository
**Moonlight Embedded running on:** Raspberry Pi
**Moonlight Embedded running on distribution:** OSMC

When I execute `moonlight quit` (to free resources after moonlight has been used and killed with Ctrl+C), it starts moonlight again instead of quiting it.

geekycow, 2017-07-21:
Forgive me for suggesting something so basic, and I don't know how things work on OSMC, but is it possible that the original moonlight task is running as a different user than the one that's issuing the `moonlight quit` command?  Have you tried sudo moonlight quit or sudo -u <user that initiated a the track> moonlight quit`?  (Sorry if I'm being patronising.)

irtimmer, 2017-08-06:
The quit command will quit the game on the host and not Moonlight itself. Therefore it starts Moonlight to find and connect to the host to be able to quit the game.

closed 2017-08-06 by irtimmer