moonlight-stream/moonlight-embedded #536: Xbox One Controller has wrong mapping

KN4CK3R, 2017-07-22:
**_NVidia Geforce Experience version:_** 3.7.0.81
**Moonlight Embedded version:** 2.4.2
**Moonlight Embedded running on:** Raspberry Pi 3
**Moonlight Embedded running on distribution:** Raspbian / Retropie

**What is the expected result?**
Correct mapping of buttons.

**What happens instead of that?**
The mapping of my Xbox One Controller (030000005e040000dd02000003020000) is wrong. The arrow keys seem to work but all other buttons are like random. R2 is the Start button for example. In Rocket League (started from Steam) the car automaticly drives backwards without pressing a button. Same happens with GTA V (not connected to Steam).

vsappel, 2017-07-28:
Same here, any fix????

KN4CK3R, 2017-07-29:
Should I add logs or something?

Gudui, 2017-08-05:
@KN4CK3R @vsappel 

I've made a fix for Debian Jessie which solves the issue.
I think you have to uninstall moonlight-embedded first, and then execute:
sudo dpkg -i ./moonlight-embedded_2.4.2-1jessie1_armhf.deb

to install it with the latest two commits.

[moonlight-embedded_2.4.2-1jessie1_armhf.zip](https://github.com/irtimmer/moonlight-embedded/files/1202473/moonlight-embedded_2.4.2-1jessie1_armhf.zip)

KN4CK3R, 2017-08-06:
The file you have provided doesn't help. Just tested it with Rocket League on Steam and the car is still driving backwards without controller interaction. Are there moonlight config files I should delete too?

blacklabel8829, 2017-08-13:
Same issue here, installed moonlight to Retropie and X-box One pad (firmware 2015) (030000005e040000dd02000003020000) has incorrect mapping. Right trigger is Start, right analog up/down is mapped to left/right. Completely unusable.

irtimmer, 2017-08-16:
This issue is a duplicate of #521. RetroPie needs custom mappings as it changes the codes used by the controller. (see also https://github.com/irtimmer/moonlight-embedded/issues/528#issuecomment-313638232)

closed 2017-08-16 by irtimmer

KN4CK3R, 2017-08-16:
Is there an easy way to generate a new mapping because the old version of moonlight-embedded was working nice (before gamecontrollerdb.txt was introduced).

lionking03, 2017-08-25:
not the better way to solve this problem but as sayed by irtimmer in [https://github.com/irtimmer/moonlight-embedded/issues/528#issuecomment-313638232](url), retropie change triggers behaviour in xpad driver
if you don't need trigger to acts as button in retropie emulator you could just make trigger act as trigger by editing /etc/modprobe.d/xpad.conf en change =1 to =0 then restart

xpad driver should work fine with moonlight after that

eberfreitas, 2017-09-23:
Has anyone succeeded at solving this? I can't really put things together from this thread in order to make it work... I have the exact same problem with the exact same game (Rocket League).

KN4CK3R, 2017-10-08:
I compiled the last version with the manual mapping:
```
sudo apt-get install libopus0 libexpat1 libasound2 libudev0 libavahi-client3 libcurl3 libevdev2 libenet7 libraspberrypi0
sudo apt-get install libssl-dev libopus-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libevdev-dev libexpat1-dev libpulse-dev uuid-dev libenet-dev cmake gcc g++ libraspberrypi-dev
wget https://github.com/irtimmer/moonlight-embedded/releases/download/v2.3.0/moonlight-embedded-2.3.0.tar.xz
tar xf moonlight-embedded-2.3.0.tar.xz
cd moonlight-embedded-2.3.0/
mkdir build
cd build/
cmake ../
make
make install

moonlight map controller.map
moonlight pair ...
```
If you get a compilation error because you need to set the C standard, add `set (CMAKE_C_STANDARD 99)` to the main CMakeLists.txt. If you get an error because the libgamestream.so can't be found, run `sudo ldconfig`.

With this older version moonlight works again for me without problems.

cg-david, 2017-11-23:
Hey everyone. 
I had the same problem: RT being Pause and other buttons where mixed up. I answered in #521 please refer to this. Hope this will solve your problem without downgrading moonlight. 
Cheers

stevoisiak, 2018-09-09:
I'm getting the same issue with an Xbox 360 controller on Raspbian Stretch. RT is being interpreted as "Pause" when ingame.