moonlight-stream/moonlight-embedded #539: Unable to build on Debian Jessie (Orange Pi)

endlessmind, 2017-08-09:
First, you have no forum so I have to post it here!
So this is not an issue with the moonlight-embedded project, but it is an issue with the documentation of it.
It's said that it's possible to compile it for Orange Pi (Lite in my case), but there is almost no information to be found on this (Googled for hours).

I've complied vdpau and ffmpeg from source, I've compiled (if needed) and installed all the package I could find any mention of that might be needed. I've followed some half-baked tutorial on this and I still get "no video output available" when trying to compile Moonlight-Embedded.
I've nowhere else to turn except here.

I pulled the latest build from github 2 hours ago, I did a fresh install of the latest Debian from armbian.com.
Latest vdpau and ffmpeg.

[logfile.txt](https://github.com/irtimmer/moonlight-embedded/files/1209915/logfile.txt)

irtimmer, 2017-08-09:
For the VDPAU to work you need glesv2, egl, vdpau, x11, libavutil and libavcodec (part of FFMpeg still missing). According to your logs you have installed the Libav version (fork of FFMpeg) of libavutil-dev and is the your source compiled FFMpeg not available. You need to correctly install FFMpeg to get Moonlight compiled or try to get it work with Libav by installing libavcodec-dev

closed 2017-08-09 by irtimmer

endlessmind, 2017-08-09:
Thanks!
I checked the ffmpeg, and I got an error on ffprobe_g (recipe error) but it continued and installed all the other parts. 
I followed [these instructions l](https://forum.armbian.com/index.php?/topic/837-ffmpeg-on-orange-pi-pc/) for compiling ffmpeg for OrangePi, maybe I missed something?

Is there any downside to using libavcodec-dev over ffmpeg?

endlessmind, 2017-08-09:
I get error in ffmpeg.c line 32: libswscale/swscale.h no such file or directory.
I'm guessing that it has to do with the ffprome_g error.
Either I need better instructions on how to compile ffmpeg or just more information in general i suppose.

irtimmer, 2017-08-10:
Swscale was used in previous versions of Moonlight. But although its no longer used the reference wasn't removed. This has been fixed with commit d7e6313a555651082380ad14c703ece16fa41da6 in master

endlessmind, 2017-08-12:
Great!
I pull the latest source now and I'm getting error
AVCodecContext has no memeber name hw_device_ctx" in ffmpeg-vaapi.c
But am I not to use vdpau, has vaapi is an x86 thing right?
I've checked, and I got the #ifdef in x11

I have attached both the "cmake ../" log and the "make" log
[build_log.txt](https://github.com/irtimmer/moonlight-embedded/files/1219715/build_log.txt)
[cmake_log.txt](https://github.com/irtimmer/moonlight-embedded/files/1219714/cmake_log.txt)

irtimmer, 2017-08-12:
Remove libva-dev to ensure it won't try to build vaapi support (which isn't supported by libav)

endlessmind, 2017-08-12:
Tried that, also tried to just comment out set(VA_ACCEL_FOUND TRUE) in the CMakeFiles file.
Both gave me "cannot find -lglesv2" error
Not compiling latest libav from source to be sure.

Starting to feel like I made some stupid mistake somewhere along the line.

endlessmind, 2017-08-13:
Compiling Cedrus and Vdpau form source seams to have fixed that, but I now get:
Unknown yupe name "VdpPresentationQueueTargetCreateX11"
'VDP_FUNC_ID_PRESENTATION_QUEUE_TARGET_CREATE_X11' undeclreated
and 
"vdp_presentation_queue_target_create_x11" is not a function or function pointer

This is with libva-dev removed

endlessmind, 2017-08-15:
Due to the absent of answers I'm guessing that it has something to do with my kernel version.
 I moved from Debian Jessie to Ubuntu Xenial due to the OS stopped booting.