moonlight-stream/moonlight-embedded #540: Returning RTP packet queued for too long

Furno221, 2017-08-11:
**_NVidia Geforce Experience version: 3.8.0.89**
**Moonlight Embedded version: newest
Moonlight Embedded running on: Raspberry Pi 3
Moonlight Embedded running on distribution:Raspbian

Output of Moonlight Embedded:
moonlight stream
Searching for server...
Connect to xxx.xxx.x.xxx...
No mapping available for Microsoft X-Box One pad (030000005e040000d102000003020000) on /dev/input/event2
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 35475, but got 35476)
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
Stream showing up on TV
**What happens instead of that?**
Blackscreen

irtimmer, 2017-08-16:
Is there any more information you can provide that might be useful? The RTP packet queued for too long is an audio packet and has nothing to do with the video.  And the other messages came from clossing Moonlight.

simper, 2017-08-19:
Similar situation here, the video and control is working, but no audio directed to Raspberry Pi output.
NVidia Geforce Experience version: 3.8.0.89
Moonlight Embedded version: 2.4.3
Moonlight Embedded running on: Raspberry Pi 2
Moonlight Embedded running on distribution: Ubuntu Mate

Output of Moonlight Embedded:
xxx@xxx-unbuntu-mate:~$ moonlight stream xxx.xxx.xxx.xxx -remote -1080 -fps 60 -app mstsc -platform pi
Connect to xxx.xxx.xxx.xxx...
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 32760, but got 32761)
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

closed 2017-08-19 by Furno221