moonlight-stream/moonlight-embedded #547: Failed to sign data

bodanu, 2017-08-24:
Hello i have:
NVidia Geforce Experience version:_3.9.0.61
Moonlight Embedded version:2.4.3
From http://archive.itimmer.nl/raspbian/moonlight jessie main
Moonlight Embedded running on:** _Raspberry Pi 3
**Moonlight Embedded running on raspbian

Whenever i try to pair with my computer i get the following:
Failed to conect to server. Failed to sign data
What am I doing wrong?

irtimmer, 2017-08-25:
That isn't supposed to happen. Try to clean $XDG_CACHE_DIR/moonlight or ~/.cache/moonlight , maybe the keys got corrupted.

badmikey83, 2017-08-25:
how can i do this?

Unorthodactyl, 2017-08-25:
Are you using raspbian stretch? Try running it with sudo

badmikey83, 2017-08-25:
yes i do. what´s the command? sorry i´m new

bodanu, 2017-08-25:
Can someone help me how should i do this? Iam really a noob with CLI.
Thank you.

bodanu, 2017-08-25:
Never mind i did it :) now it works fine thank you so much !

closed 2017-08-25 by irtimmer