moonlight-stream/moonlight-embedded #548: Can´t install on Debian Stretch on Raspberry Pi 2/3

badmikey83, 2017-08-25:
Please provide the following info.

**_NVidia Geforce Experience version:_*newest*
**Moonlight Embedded version: 2.4.0 (?)
**Moonlight Embedded source: http://archive.itimmer.nl/moonlight/raspian jessie main
**Moonlight Embedded running on:** _Raspberry Pi 2+3..._
**Moonlight Embedded running on distribution: Devian Stretch

**Verbose output `-verbose` of Moonlight Embedded:**
     can´t install because of libssl1.0.0 dependencie....
**What is the expected result?**
Install Moonlight-embedded
**What happens instead of that?**
Errors and no moonlight

Unorthodactyl, 2017-08-25:
Raspbian Stretch only just released. You can download and install libsll1.0.0 from here https://packages.debian.org/jessie/armhf/libssl1.0.0/download
then you will have to pair and stream with sudo

badmikey83, 2017-08-25:
Thanks. That´s what i tried. It worked, but than i got the "Can´t Pair" Problem, but it all worked out

closed 2017-08-25 by badmikey83